platformio/platform-espressif32 #471: Add support for ESP-IDF v4.3

silikone, 2021-01-21:
Missing support for newer versions of the IDF tools, e.g. 4.2 and 4.3 (although still not released)

Would be totally fine if the support means dropping the Arduino features, or this was broken into different support packages.

meltdown03, 2021-01-26:
It's released now!

kind3r, 2021-02-15:
Would be great to have those new websocket features :D at least in the dev branch for the time being.

GeneKong, 2021-03-05:
How to get the ESP-IDF version number, and how to use self defined version for office github?

w4rum, 2021-03-11:
Is there a way that I can manually hack the latest esp-idf version in?

I'm running into [an issue](https://github.com/espressif/esp-idf/issues/6369) that seems to have been fixed via [this commit](https://github.com/espressif/esp-idf/commit/f0b0418ed412c94114f0ed9176494f3e2c19a9a9) and (I think) has been included in 4.3-beta1.

I tried using
```
platform_packages =
    framework-espidf @ https://github.com/espressif/esp-idf.git
```
and that causes the latest revision to be pulled from git when running `pio run -t upload`. But after pulling, `~/.platformio/packages/framework-espidf` doesn't contain a lot of files and the build process fails with
```
Reading CMake configuration...
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring incomplete, errors occurred!
See also "[... project dir ...]/.pio/build/esp32cam/CMakeFiles/CMakeOutput.log".

CMake Error at CMakeLists.txt:2 (include):
  include could not find load file:

    [... home ...]/.platformio/packages/framework-espidf/tools/cmake/project.cmake

```
Directory listing of `~/.platformio/packages/framework-espidf`:
```
total 64K
drwxr-xr-x  2 tim users 4.0K Mar 11 16:11 .
drwxr-xr-x 12 tim users 4.0K Mar 11 16:11 ..
-rw-r--r--  1 tim users 2.6K Mar 11 16:11 cjson-utils-tests.json
-rw-r--r--  1 tim users  183 Mar 11 16:11 .editorconfig
-rw-r--r--  1 tim users   34 Mar 11 16:11 .gitignore
-rw-r--r--  1 tim users   25 Mar 11 16:11 .npmignore
-rw-r--r--  1 tim users  393 Mar 11 16:11 package.json
-rw-r--r--  1 tim users  226 Mar 11 16:11 .piopm
-rw-r--r--  1 tim users 2.3K Mar 11 16:11 README.md
-rw-r--r--  1 tim users 4.0K Mar 11 16:11 spec_tests.json
-rw-r--r--  1 tim users  17K Mar 11 16:11 tests.json
-rw-r--r--  1 tim users   17 Mar 11 16:11 version.txt
```

w4rum, 2021-03-14:
Alright, I've found a way to hack the latest version in. It's not pretty:

1. In `~/.platformio/platforms/espressif32/platform.json`, change the version of `framework-espidf`:
    ```json
    "framework-espidf": {
      "type": "framework",
      "optional": true,
      "owner": "platformio",
      "version": "https://github.com/espressif/esp-idf.git"
    },
    ```
2. Try to compile your project. This will pull the repository and all submodules *and then fail*.
    ```fish
    > pio run

    Reading CMake configuration...
    -- Configuring incomplete, errors occurred!

    CMake Error at CMakeLists.txt:2 (include):
    include could not find load file:

        ~/.platformio/packages/framework-espidf/tools/cmake/project.cmake

    === [FAILED] Took 78.09 seconds ===
    ```
3. Download the esp-idf repo manually:
    ```fish
    git clone -b v4.3-beta1 --recursive https://github.com/espressif/esp-idf.git esp-idf-v4.3-beta1
    ```
4. Copy everything over to the espidf repo created by pio. (Note: I'm using fish syntax here.)
    ```fish
    cp -r esp-idf-v4.3-beta1/{.,}* ~/.platformio/packages/framework-espidf/
    ```
5. `cd ~/.platformio/packages/framework-espidf/`
6. Install dependencies. (Note: Arch Linux version, check the [docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#step-1-install-prerequisites) for your system.)
    ```fish
    sudo pacman -S --needed gcc git make flex bison gperf python-pip cmake ninja ccache dfu-util libusb
    ```
7. Run the install script. (Again, use the version for you shell if you're not using fish.)
    ```fish
    ./install.fish
    ```
8. Source the export file:
    ```fish
    IDF_PATH=~/.platformio/packages/framework-espidf/ source export.fish
    ```
9. `cd` back into your project directory.
10. Try building your project with ESP-IDF's `idf.py`. Note that this will probably fail on the last step but still generate what's necessary to build with `pio`.
    ```fish
    IDF_PATH=~/.platformio/packages/framework-espidf/ ~/.platformio/packages/framework-espidf/tools/idf.py build
    ```
11. Build and upload with `pio` as usual:
    ```fish
    pio run -t upload
    ```

amaschas, 2021-05-11:
Any movement on this? I keep running into documentation online for ESP-IDF, and then halfway through realizing it refers to 4.3 and that I can't use platform.

maxgerhardt, 2021-06-16:
>Any movement on this? I keep running into documentation online for ESP-IDF, and then halfway through realizing it refers to 4.3 and that I can't use platform.

Well to be fair you can change the version to v.4.2.1 in the official ESP-IDF documentation website. https://docs.espressif.com/projects/esp-idf/en/v4.2.1/esp32/index.html

I've forked ESP-IDF here and put a `package.json` in the `release/v4.3` branch (https://github.com/maxgerhardt/esp-idf/tree/release/v4.3). 

If you use a standard [espidf-blink](https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-blink) project and add

```ini
platform_packages =
   framework-espidf@https://github.com/maxgerhardt/esp-idf.git#release/v4.3
```
to it ([docs](https://docs.platformio.org/en/latest/projectconf/section_env_platform.html#platform-packages)) you can use it. It works just fine for me.

```
Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.2.1) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40300.160621+sha.beaf3ab
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 2.80400.210114 (8.4.0)
Reading CMake configuration...
[...]
[..]
Checking size .pio\build\esp32dev\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   3.9% (used 12620 bytes from 327680 bytes)
Flash: [==        ]  15.5% (used 162012 bytes from 1048576 bytes)
Building .pio\build\esp32dev\firmware.bin
esptool.py v3.0
========================= [SUCCESS] Took 65.98 seconds =========================
```

commit 5384f152 (2021-06-23): Initial support for ESP-IDF v4.3

Resolve #566, resolve #471, resolve #561
closed 2021-06-23 by valeros