platformio/platform-espressif32 #473: [Help Wanted] espidf - IDF v4.3 testing / x509_crt_bundle.S missing

meltdown03, 2021-01-24:
In trying to get  4.3 to work, I run into an issue with `x509_crt_bundle.S` not being found. The CMakeFiles.txt file in the mbedtls components directory should generate it but doesn't. Does the builder `espidf.py` process all the CMakeLists.txt files and run the commands in them? If I run menuconfig and turn off `CONFIG_MBEDTLS_CERTIFICATE_BUNDLE`, **_it builds fine_** which is promising for IDF 4.3 support. I just need help getting the Certificate bundle created.

If I could get a more verbose output for Platformio that might help me. I do see that the partition_table is made directly by `espidf.py` instead of using the CMakeLists.txt script. Is there a specific reason for this behavior?

Any help would be appreciated.

Note:
I used the `pio package pack .` command in the master branch of my `IDF_PATH` to make the custom framework-espidf package if anyone else is interested. You need to put a `package.json` file in the root directory first and then set `platform_packages = framework-espidf @ file:///<path-to-generated.tar.gz>` in your platformio.ini

meltdown03, 2021-01-25:
### Dirty Hack 
At the very end of https://github.com/platformio/platform-espressif32/blob/develop/builder/frameworks/espidf.py, I added 4 lines and got it to work:

``` bash
GENERATOR_PY = os.path.join(FRAMEWORK_DIR, "components", "mbedtls", "esp_crt_bundle", "gen_crt_bundle.py")
CACERT = os.path.join(FRAMEWORK_DIR, "components", "mbedtls", "esp_crt_bundle", "cacrt_all.pem")

env.Execute("%s %s --input %s" % (env.get("PYTHONEXE"), GENERATOR_PY, CACERT))
env.TxtToAsm("x509_crt_bundle",os.path.join("$PROJECT_DIR", "x509_crt_bundle"))
```

It's not perfect as it doesn't respect the `CONFIG_MBEDTLS_CERTIFICATE_BUNDLE` defines. It just loads the Full Certificate bundle.
I started by just using the `extra_scripts` and `board_build.embed_files` flags in my `platformio.ini` file.

I also used `pio package pack .` with the latest xtensa toolchain and have no issues building against the latest IDF.

meltdown03, 2021-01-26:
Fixed with 649b3816a933376f965279e21ddece65b6766997

closed 2021-01-26 by meltdown03

FireDeveloper, 2021-01-30:
Can you please upload the package.json that you used?
I am also interested in using later idf packages.
Thank you

meltdown03, 2021-01-30:
```bash
{
  "name": "framework-espidf",
  "version": "3.40300.210127",
  "description": "Espressif IoT Development Framework. Official development framework for ESP32 chip",
  "keywords": [
    "framework",
    "esp32",
    "espressif"
  ],
  "homepage": "https://docs.espressif.com/projects/esp-idf/en/latest/esp32/",
  "license": "Apache-2.0",
  "repository": {
    "type": "git",
    "url": "https://github.com/espressif/esp-idf"
  }
}
```
I put this in my ESP_IDF path, then ran `pio package pack.` to create a tar.gz.

meltdown03, 2021-01-30:
You also have to install `kconfiglib` using the python in the `.platformio/penv/bin` directory to build 4.3 projects