platformio/platform-espressif32 #474: (Develop Branch) Build failing ESP32 S2 missing file

FireDeveloper, 2021-01-25:
Processing esp32dev (platform: https://github.com/platformio/platform-espressif32.git; board: esp32-s2-saola-1; framework: espidf)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Platform Manager: espressif32 @ 2.1.0+sha.ab87490 is already installed
The platform 'https://github.com/platformio/platform-espressif32.git' has been successfully installed!
The rest of the packages will be installed later depending on your build environment.
Tool Manager: Installing platformio/toolchain-xtensa32s2 @ ~1.80400.0
Unpacking  [####################################]  100%
Tool Manager: toolchain-xtensa32s2 @ 1.80400.210114 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s2-saola-1.html
PLATFORM: Espressif 32 (2.1.0+sha.ab87490) > Espressif ESP32-S2-Saola-1
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, esp32s2_kaluga_v1, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40200.210118 (4.2.0)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-idf 1.0.1 
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32s2ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32s2 1.80400.210114 (8.4.0)
Reading CMake configuration...
Generating assembly for certificate bundle...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\esp32dev\src\main.o
Compiling .pio\build\esp32dev\tinyusb\port\common\src\descriptors_control.o
Compiling .pio\build\esp32dev\tinyusb\port\common\src\usb_descriptors.o
Compiling .pio\build\esp32dev\tinyusb\port\common\src\usbd.o
C:\users\fire\.platformio\packages\framework-espidf\components\tinyusb\port\common\src\usb_descriptors.c:16:10: fatal error: usb_descriptors.h: No such file or directory

*************************************************************************
* Looking for usb_descriptors.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:usb_descriptors.h"
* Web  > https://platformio.org/lib/search?query=header:usb_descriptors.h
*
*************************************************************************

 #include "usb_descriptors.h"
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
C:\users\fire\.platformio\packages\framework-espidf\components\tinyusb\port\common\src\descriptors_control.c:15:10: fatal error: descriptors_control.h: No such file or directory

*****************************************************************************
* Looking for descriptors_control.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:descriptors_control.h"
* Web  > https://platformio.org/lib/search?query=header:descriptors_control.h
*
*****************************************************************************

 #include "descriptors_control.h"
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
C:\users\fire\.platformio\packages\framework-espidf\components\tinyusb\port\common\src\usbd.c:28:10: fatal error: tusb_option.h: No such file or directory

*********************************************************************
* Looking for tusb_option.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:tusb_option.h"
* Web  > https://platformio.org/lib/search?query=header:tusb_option.h
*
*********************************************************************

 #include "tusb_option.h"
          ^~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32dev\tinyusb\port\common\src\usb_descriptors.o] Error 1
*** [.pio\build\esp32dev\tinyusb\port\common\src\descriptors_control.o] Error 1
*** [.pio\build\esp32dev\tinyusb\port\common\src\usbd.o] Error 1
================================================================================================ [FAILED] Took 166.42 seconds ================================================================================================
The terminal process "C:\Users\Fire\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

valeros, 2021-01-25:
Hi @FireDeveloper ! Thanks for reporting. Have you enabled USB in menuconfig (for example as in this example https://github.com/platformio/platform-espressif32/blob/develop/examples/espidf-peripherals-usb/sdkconfig.defaults) ? Maybe you could put together a minimal example to reproduce the issue?

FireDeveloper, 2021-01-25:
I didn't configure anything, just an empty project with this in platformio.ini
As far I can tell idf builds everything and then chooses which parts to link.

[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32-s2-saola-1
framework = espidf

The file that is reported as missing exists at path
C:\users\fire.platformio\packages\framework-espidf\components\tinyusb\port\common\include\usb_descriptors.h
the error is at file
C:\users\fire.platformio\packages\framework-espidf\components\tinyusb\port\common\src\usb_descriptors.c

Could be a missing configuration for "include" subfolder search?

Also something that looks odd is that there is a path:
C:\Users\Fire\.platformio\packages\framework-espidf\components\tinyusb\port\esp32s2\src
Maybe it should have build 'esp32s2' folder and not 'common'
 [RGB_S2.zip](https://github.com/platformio/platform-espressif32/files/5868733/RGB_S2.zip)

FireDeveloper, 2021-01-25:
Its probably an error in platform io scripts
An empty ESP32 project won't build either with similar error.

Executing task in folder TC100_ESP32: C:\Users\Fire\.platformio\penv\Scripts\pio.exe run <

Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40100.200827 (4.1.0)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 2.80200.200827 (8.2.0)
Reading CMake configuration...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\esp32dev\esp-idf\src\main_esp_idf.c.o
Generating LD script .pio\build\esp32dev\esp32_out.ld
Generating partitions .pio\build\esp32dev\partitions.bin
Compiling .pio\build\esp32dev\esp-idf\app_trace\app_trace.c.o
Compiling .pio\build\esp32dev\esp-idf\app_trace\app_trace_util.c.o
Compiling .pio\build\esp32dev\esp-idf\app_trace\host_file_io.c.o
C:\users\fire\.platformio\packages\framework-espidf\components\app_trace\app_trace.c:159:10: fatal error: soc/soc.h: No such file or directory
 #include "soc/soc.h"
          ^~~~~~~~~~~
compilation terminated.
In file included from src\main_esp_idf.c:12:
C:/users/fire/.platformio/packages/framework-espidf/components/freertos/include/freertos/FreeRTOS.h:101:10: fatal error: freertos/FreeRTOSConfig.h: No such file or directory
 #include "freertos/FreeRTOSConfig.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from C:\users\fire\.platformio\packages\framework-espidf\components\app_trace\app_trace_util.c:15:
C:\users\fire\.platformio\packages\framework-espidf\components\freertos\include/freertos/FreeRTOS.h:101:10: fatal error: freertos/FreeRTOSConfig.h: No such file or directory
 #include "freertos/FreeRTOSConfig.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32dev\esp-idf\app_trace\app_trace.c.o] Error 1
*** [.pio\build\esp32dev\esp-idf\src\main_esp_idf.c.o] Error 1
*** [.pio\build\esp32dev\esp-idf\app_trace\app_trace_util.c.o] Error 1
In file included from C:\users\fire\.platformio\packages\framework-espidf\components\app_trace\include/esp_app_trace_util.h:21,
                 from C:\users\fire\.platformio\packages\framework-espidf\components\app_trace\include/esp_app_trace.h:19,
                 from C:\users\fire\.platformio\packages\framework-espidf\components\app_trace\host_file_io.c:26:
C:\users\fire\.platformio\packages\framework-espidf\components\freertos\include/freertos/FreeRTOS.h:101:10: fatal error: freertos/FreeRTOSConfig.h: No such file or directory
 #include "freertos/FreeRTOSConfig.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32dev\esp-idf\app_trace\host_file_io.c.o] Error 1
================================================================================================= [FAILED] Took 4.21 seconds =================================================================================================
The terminal process "C:\Users\Fire\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

(.ini file)

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf

valeros, 2021-01-25:
Thanks for reporting, please update to the latest revision from the dev branch and try again.

commit 14e3f31d (2021-01-25): Build tinyusb lib only when it's explicitly enabled

Resolve #474
closed 2021-01-25 by valeros