platformio/platform-espressif32 #475: TTGO LilyGO T-ETH-POE support

Mingyu-Kim, 2021-01-25:
Can you add TTGO's T-ETH-POE support?

Board info
https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-POE
http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=1307&FId=t3:50033:3

Example
https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-POE/tree/master/example/eth

Thank you

cnoork, 2021-02-18:
I would also like that the board is added to PlatformIO

Lgeorge-V, 2024-01-20:
I would greatly appreciate it if this board was added to PlatformIO.

WombatHollow, 2024-01-20:
Yes please to adding ttgo boards including this one to platformio.   PLEASE

PurP13cray0n, 2024-01-20:
Please add this board to the list of boards supported

valeros, 2024-01-22:
There is no proper variant file for this board in the official Arduino core for ESP32 at the moment, so it needs to be added to this repo https://github.com/espressif/arduino-esp32 in the first place.

PurP13cray0n, 2024-01-22:
Thank you for directing me towards the underlying problem, I have made a feature request in the Official Arduino core for ESP32.
[https://github.com/espressif/arduino-esp32/issues/9159](url)