platformio/platform-espressif32 #478: Use update openocd configuration files

maxgerhardt, 2021-01-29:
When debugging with

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

upload_protocol = esp-prog
debug_tool = esp-prog
debug_init_break = tbreak setup
```

The board definition says

https://github.com/platformio/platform-espressif32/blob/883518fa60efc9738f53f6743216be435e5ff1fe/boards/esp32dev.json#L20-L22 

but further the [openocd config says](https://github.com/espressif/openocd-esp32/blob/a6f77482e392c0cc7bb06d0f9b0310962ac7bebe/tcl/board/esp-wroom-32.cfg#L1-L3)

```
echo "WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future release."
set ESP32_FLASH_VOLTAGE 3.3
source [find target/esp32.cfg]
```
so users will receive a 

```
*WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future release*.
```
warning which might confuse them. 

It would be good to update the `openocd_board` value and maybe add a additional parameter to set the flash voltage, so that it's also nicely and independenly configurable I think.

Bobobel, 2024-10-27:
Where is that "enhancement"? 
The warning still apears to date!

Jason2866, 2024-10-27:
The feature request is still open and not implemented