platformio/platform-espressif32 #480: Linking error after upgraded from pio espressif 2.1.0 to 3.0.0

bheestandLMT, 2021-02-01:
Had a project that compiled fine with PIO and platform=espressif32 version 2.1.0.
Upgraded to espressif32 version 3.0.0.
Now I am getting a linking error. It appears to be due to my OTA ca_cert file (located in server_certs/ca_cert.pem)

`> Executing task: C:\Users\bheestand\.platformio\penv\Scripts\pio.exe run <

Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.0.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40200.210118 (4.2.0)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0) 
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 2.80400.210114 (8.4.0)
Reading CMake configuration...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Linking .pio\build\esp32dev\firmware.elf
c:/users/bheestand/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\src\app_ota.o:(.literal.ota_task+0x18): undefined reference 
to `_binary_ca_cert_pem_start'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
============================================================================================== [FAILED] Took 37.84 seconds ==============================================================================================
The terminal process "C:\Users\bheestand\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.`

Any help would be appreciated.

valeros, 2021-02-01:
In case with the latest ESP-IDF v4.2, your certificate file also need to be specified in `CMakeLists.txt` using the target_add_binary_data function, for example as it done here https://github.com/platformio/platform-espressif32/blob/develop/examples/espidf-aws-iot/src/CMakeLists.txt

bheestandLMT, 2021-02-01:
Here is my CMakeLists.txt:
`cmake_minimum_required(VERSION 3.16.0)
set(EXTRA_COMPONENTS_DIRS components/)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(smart-gateway-voice-control)`

If I add:
`target_add_binary_data(${COMPONENT_TARGET} "server_certs/ca_cert.pem" TEXT)`

I get:
`CMake Error at CMakeLists.txt:5 (target_add_binary_data):
  target_add_binary_data Function invoked with incorrect arguments for
  function named: target_add_binary_data`

In advance, thanks for your help!

valeros, 2021-02-01:
That line `target_add_binary_data(${COMPONENT_TARGET} "server_certs/ca_cert.pem" TEXT)` should be added to `src/CMakeLists.txt`

bheestandLMT, 2021-02-01:
thank you @valeros ! Works!
I appreciate you!

meltdown03, 2021-02-02:
Is `set(EXTRA_COMPONENTS_DIRS components)` correct and necessary?
I seem to have to use `list(APPEND EXTRA_COMPONENT_DIRS <path/to/esp-rainmaker/NOT/IN/PROJECT/DIR> <another-component>)`
for esp-rainmaker projects

valeros, 2021-02-02:
@meltdown03 https://docs.platformio.org/en/latest/frameworks/espidf.html#esp-idf-components

closed 2021-02-02 by valeros