platformio/platform-espressif32 #486: Boards have incorrect maximum_ram_size in json files

meltdown03, 2021-02-06:
The normal ESP32 should have `532480` bytes SRAM. The `az-delivery-devkit-v4` seems to be correct but the others using the same module are only `327680`.

https://github.com/platformio/platform-espressif32/blob/55f50dc8efe7ffca6cdbd82c6f5e22b21d21bebf/boards/az-delivery-devkit-v4.json#L30

https://github.com/platformio/platform-espressif32/blob/55f50dc8efe7ffca6cdbd82c6f5e22b21d21bebf/boards/esp32dev.json#L30

valeros, 2021-02-08:
We use the same size as the Arduino IDE https://github.com/espressif/arduino-esp32/blob/master/boards.txt#L21

closed 2021-02-08 by valeros

carl3, 2022-02-08:
The boards files ttgo-t-beam.json, ttgo-t7-v13-mini32.json, and ttgo-t7-v14-mini32.json have an incorrect maximum_ram_size of 1310720. It should be 327680, similar to all the other boards. It is 327680 in the  https://github.com/espressif/arduino-esp32/blob/master/boards.txt