platformio/platform-espressif32 #487: See Ethernet is not present in example

giorgio80, 2021-02-07:
try to run the ethernet example from esp-idf with no success. 

Could a clear example be included ?

valeros, 2021-02-08:
Hi @giorgio80 ! Could you please provide more details? What exactly doesn't work? What is the error message?

giorgio80, 2021-02-08:
sure.   I'm using the example provided here from Espressif: 

https://github.com/espressif/esp-idf/tree/master/examples/ethernet

Actually, in menuconfig, the option for example configuration are not present ( or I'm miss it ?  ) and this results in compilation errors 
I'm wondering if there's a way to force this options.

valeros, 2021-02-08:
So, what's the error message exactly?

giorgio80, 2021-02-09:
here my output: 

Those variables should be included when adding the config examples as per example explanation. unfortunately I cannot find "config example" in menuconfig: 

`In the Example Configuration menu:

Choose the kind of Ethernet under Ethernet Type.
If Internal EMAC is selected:
Choose PHY device under Ethernet PHY Device, by default, the ESP32-Ethernet-Kit has an IP101 on board.
Set GPIO number used by SMI signal under SMI MDC GPIO number and SMI MDIO GPIO number respectively.
If DM9051 Module is selected:
Set SPI specific configuration, including SPI host number, GPIO number and clock rate.
Set GPIO number used by PHY chip reset under PHY Reset GPIO number, you may have to change the default value according to your board schematic. PHY hardware reset can be disabled by set this value to -1.
Set PHY address under PHY Address, you may have to change the default value according to your board schematic.`

`
src/main.c: In function 'app_main':
src/main.c:77:27: error: 'CONFIG_EXAMPLE_ETH_PHY_ADDR' undeclared (first use in this function)
     phy_config.phy_addr = CONFIG_EXAMPLE_ETH_PHY_ADDR;
                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~
src/main.c:77:27: note: each undeclared identifier is reported only once for each function it appears in
src/main.c:78:33: error: 'CONFIG_EXAMPLE_ETH_PHY_RST_GPIO' undeclared (first use in this function); did you mean 'CONFIG_CONSOLE_UART_RX_GPIO'?
     phy_config.reset_gpio_num = CONFIG_EXAMPLE_ETH_PHY_RST_GPIO;
                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                 CONFIG_CONSOLE_UART_RX_GPIO
src/main.c:98:24: error: 'CONFIG_EXAMPLE_ETH_SPI_MISO_GPIO' undeclared (first use in this function); did you mean 'CONFIG_CONSOLE_UART_RX_GPIO'?
         .miso_io_num = CONFIG_EXAMPLE_ETH_SPI_MISO_GPIO,
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                        CONFIG_CONSOLE_UART_RX_GPIO
src/main.c:99:24: error: 'CONFIG_EXAMPLE_ETH_SPI_MOSI_GPIO' undeclared (first use in this function); did you mean 'CONFIG_CONSOLE_UART_RX_GPIO'?
         .mosi_io_num = CONFIG_EXAMPLE_ETH_SPI_MOSI_GPIO,
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                        CONFIG_CONSOLE_UART_RX_GPIO
src/main.c:100:24: error: 'CONFIG_EXAMPLE_ETH_SPI_SCLK_GPIO' undeclared (first use in this function); did you mean 'CONFIG_ETH_RMII_CLK_IN_GPIO'?
         .sclk_io_num = CONFIG_EXAMPLE_ETH_SPI_SCLK_GPIO,
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                        CONFIG_ETH_RMII_CLK_IN_GPIO
In file included from /Users/giorgiocasti/.platformio/packages/framework-espidf/components/esp_timer/include/esp_timer.h:44,
                 from /Users/giorgiocasti/.platformio/packages/framework-espidf/components/freertos/xtensa/include/freertos/portmacro.h:84,
                 from /Users/giorgiocasti/.platformio/packages/framework-espidf/components/freertos/include/freertos/portable.h:95,
                 from /Users/giorgiocasti/.platformio/packages/framework-espidf/components/freertos/include/freertos/FreeRTOS.h:107,
                 from src/main.c:3:
src/main.c:104:40: error: 'CONFIG_EXAMPLE_ETH_SPI_HOST' undeclared (first use in this function); did you mean 'CONFIG_ESP_TASK_WDT'?
     ESP_ERROR_CHECK(spi_bus_initialize(CONFIG_EXAMPLE_ETH_SPI_HOST, &buscfg, 1));
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/giorgiocasti/.platformio/packages/framework-espidf/components/esp_common/include/esp_err.h:118:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
In file included from /Users/giorgiocasti/.platformio/packages/framework-espidf/components/esp_eth/include/esp_eth_netif_glue.h:16,
                 from /Users/giorgiocasti/.platformio/packages/framework-espidf/components/esp_netif/include/esp_netif.h:26,
                 from src/main.c:5:
src/main.c:137:50: error: 'mac' undeclared (first use in this function)
     esp_eth_config_t config = ETH_DEFAULT_CONFIG(mac, phy);
                                                  ^~~
/Users/giorgiocasti/.platformio/packages/framework-espidf/components/esp_eth/include/esp_eth.h:97:16: note: in definition of macro 'ETH_DEFAULT_CONFIG'
         .mac = emac,                     \
                ^~~~
src/main.c:137:55: error: 'phy' undeclared (first use in this function)
     esp_eth_config_t config = ETH_DEFAULT_CONFIG(mac, phy);
                                                       ^~~
/Users/giorgiocasti/.platformio/packages/framework-espidf/components/esp_eth/include/esp_eth.h:98:16: note: in definition of macro 'ETH_DEFAULT_CONFIG'
         .phy = ephy,                     \
                ^~~~
src/main.c:96:25: warning: unused variable 'spi_handle' [-Wunused-variable]
     spi_device_handle_t spi_handle = NULL;
                         ^~~~~~~~~~
src/main.c:76:22: warning: variable 'phy_config' set but not used [-Wunused-but-set-variable]
     eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();
                      ^~~~~~~~~~
src/main.c:75:22: warning: unused variable 'mac_config' [-Wunused-variable]
     eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();`

valeros, 2021-02-11:
Could you please attach a minimal example to reproduce the issue?

giorgio80, 2021-02-14:
hi @valeros , 

simply run the example from espressif here (there's full code and explain): 

https://github.com/espressif/esp-idf/tree/master/examples/ethernet

I'm not able to make it running at all

valeros, 2021-02-15:
That links shows four different examples, all of them don't work?

giorgio80, 2021-02-28:
exactly , all missing the ETHERNET example configuration should be activated from menuconfig . 
I can see it in esp-idf menuconfig but not in platformio menuconfig option. 

are you able to run the BASIC example ?

daverathbone, 2022-03-08:
it's still broken. ...
The ESP-IDF path for esp_eth is not  part of the pio build search  it needs the following from the real  ESP-IDF lib
esp-idf-v4.4\components\hal\include\hal\eth_types.h
esp-idf-v4.4\components\esp_eth\include\esp_eth_mac.h
esp-idf-v4.4\components\esp_eth\include\esp_eth_phy.h
esp-idf-v4.4\components\esp_eth\include\eth_phy_regs_struct.h
esp-idf-v4.4\components\esp_eth\include\esp_eth_com.h
Not to forget the actual Ethernet chip includes and C source which is in
esp-idf-v4.4\components\esp_eth\src
which is in my case "esp_eth_phy_lan87xx.c" for my chip

Why platformio could not be a wrapper  from Espressif  library rather than trying to push it into ".platformio" 
only the novice esp32 wire functions for Arduino huggers  need be in the PlatformIO tool 
As the ini file with compiler reductives, paths and includes are a real cross framework mashup.

PlatformIO is mainly for crossdressed ESP32 development for Arduino to ESP32.
But not for production development tool. 

Note ethernet 
All works with  ESP_IDF py tools or adding the  ESP IDF extension to Visual studio

closed 2022-08-10 by stale[bot]