platformio/platform-espressif32 #489: toolchain-extensa32 - version 2.80400.0 not found for linux_x86_64

MariusVB, 2021-02-09:
As the title says i cannot upgrade the extensa32 toolchain to the latest version and get c++17 syntax/support.

The toolchain is supported on linux_amd64, which is the same as linux_x86_64. 

The release can be found here:
https://github.com/espressif/crosstool-NG/releases/download/esp-2020r3/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-linux-amd64.tar.gz

Please correct system linux_x86_64 to also include linux_amd64..

MariusVB, 2021-02-10:
Got it to download and install using these:

```
toolchain-xtensa32 @ https://github.com/Jason2866/platform-espressif32/releases/download/8.4.0/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-linux-amd64.tar.gz
toolchain-xtensa32 @ https://github.com/Jason2866/platform-espressif32/releases/download/8.2.0/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-linux-amd64.tar.gz
toolchain-xtensa32 @ 3.80200.200512
```
However, when building a project I receive several errors, such as:
```
/home/marius/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/8.4.0/xtensa-esp32-elf/no-rtti/bits/gthr-default.h: In function 'int __gthread_mutex_timedlock(__gthread_mutex_t*, const __gthread_time_t*)':
/home/marius/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/8.4.0/xtensa-esp32-elf/no-rtti/bits/gthr-default.h:781:12: error: 'pthread_mutex_timedlock' was not declared in this scope
     return __gthrw_(pthread_mutex_timedlock) (__mutex, __abs_timeout);
            ^~~~~~~~
/home/marius/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/8.4.0/xtensa-esp32-elf/no-rtti/bits/gthr-default.h:781:12: note: suggested alternative: 'pthread_mutex_trylock'
```

Looks like the same errors as in this thread:
https://community.platformio.org/t/possible-to-use-c-17/12272/9

Doing as "Kooscode" using build flags and build unflags, still doesn't work for me.

ivankravets, 2021-02-11:
Please open [PlatformIO CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and provide an output of 
```
pio system info
```

MariusVB, 2021-02-15:
[image]

valeros, 2021-02-15:
We have `linux_x86_64` packages in our CDN, please try `pio system prune -f` and try to build your project again.

MariusVB, 2021-02-15:
Which toolchain-xtensa32 version should i specify for the latest GCC with c++17 support?
` toolchain-xtensa32 @ ?? `

Do you have any way to browse versions in your CDN?

valeros, 2021-02-15:
It depends on the framework you want to use. The latest toolchains are only available with ESP-IDF, as the current stable Arduino core v1.0.4 depends on the old ESP-IDF v3.x branch. Anyway, you can try to use the latest toolchain at your own risk. `~2.80400.0` is the right specification that will download the latest packages available.

MariusVB, 2021-02-15:
And thats where the error comes in...

`Error: Could not find the package with 'platformio/toolchain-xtensa32 @ 2.80400.0' requirements for your system 'linux_x86_64'`

valeros, 2021-02-15:
You need to specify a range of accepted versions `~2.80400.0` instead of the precise version `2.80400.0`.

MariusVB, 2021-02-15:
Ah, thanks. Still getting the pthread_mutex_trylock error though. This is due to Arduino-ESP still depending on ESP-IDF v.3? Do you know what the update plan looks like?

> ```
> /home/marius/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/8.4.0/xtensa-esp32-elf/no-rtti/bits/gthr-default.h: In function 'int [...]

valeros, 2021-02-15:
>  This is due to Arduino-ESP still depending on ESP-IDF v.3?

That's correct. As an alternative you can use a special branch with the latest Arduino core based on IDF v4.2:
```ini
```

MariusVB, 2021-02-15:
Thanks!

As you can see, I'm now using the latest version 3.0.0 from Github in conjunction with extensa 2.8 (8.4):
[image]

However, I'm still getting errors while building. 
```

/home/marius/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/sys-include/stdlib.h:155:44: error: expected initializer before '__result_use_check'
 void *reallocarray(void *, size_t, size_t) __result_use_check __alloc_size(2)
                                            ^~~~~~~~~~~~~~~~~~
/home/marius/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/sys-include/stdlib.h:340:52: error: expected initializer before '__alloc_align'

/home/marius/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/8.4.0/xtensa-esp32-elf/no-rtti/bits/gthr-default.h: In function 'int __gthread_mutex_timedlock(__gthread_mutex_t*, const __gthread_time_t*)':
/home/marius/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/8.4.0/xtensa-esp32-elf/no-rtti/bits/gthr-default.h:781:12: error: 'pthread_mutex_timedlock' was not declared in this scope
     return __gthrw_(pthread_mutex_timedlock) (__mutex, __abs_timeout);
```

valeros, 2021-02-15:
Please see my comment above, you need not only the latest platform, but also the latest revision from the special `feature/arduino-idf-v4.2` branch. 

Anyway, since you managed to download the desired package and it doesn't because of incompatible software. I'm closing the issue as it got off topic.

closed 2021-02-15 by valeros

MariusVB, 2021-02-15:
Sorry, thought it was a comment at the end of the line.

Thanks for the help, it compiles now!

One last question, my flash size went up about 11 percentage points - from 83 to 94 - is this due to the new feature branch?

valeros, 2021-02-15:
It's hard to guess, it might be caused by several reasons, including a new toolchain and updated Arduino core.