platformio/platform-espressif32 #491: Error: Couldn't find the main target of the project!

dblock247, 2021-02-15:
I am using CLion with the Platformio plug-in. I chose the NodeMCU->NodeMCU-32S->espidf board settings. However when I try to upload a program to the development board I get the following error. 

> Error: Couldn't find the main target of the project!

I'll admit I am new to programming microcontrollers but I am a seasoned software developer. I can't even upload an empty "void app_main() {}" program. Is there a step I am missing or is this a bug that has yet to be fixed?

valeros, 2021-02-15:
Hi @dblock247 ! Starting with version 4.0, ESP-IDF uses a build system based on CMake. In order to provide more seamless integration we also switched to CMake as the source of project build configuration. Unfortunately, there is a conflict between CMakeLists.txt used by ESP-IDF and CMakeLists.txt which we generate for CLion. At the moment we're working on better integration with CLion without this intermediate "hacky" CMakeLists.txt, but there is no ETA for this feature.

closed 2021-02-15 by valeros

Azjhong, 2021-06-14:
> Hi @dblock247 ! Starting with version 4.0, ESP-IDF uses a build system based on CMake. In order to provide more seamless integration we also switched to CMake as the source of project build [...]

Has this problem been solved, now does clion support esp-idy development features

RAYs3T, 2022-01-19:
Is there a work-a-round for this to compile with CLion until a fix is available?