platformio/platform-espressif32 #492: FreeRtOS uxTaskgetState

SalsaMicha, 2021-02-16:
Hello,
can the functions uxTaskGetState being supported in next release. The functions to view running tasks are not within the Lib files.
the following defines needs to be set, when creating the FreeRTOS Lib. CONFIG_FREERTOS_USE_TRACE_FACILITY, CONFIG_FREERTOS_USE_STATS_FORMATTING_FUNCTIONS, CONFIG_FREERTOS_GENERATE_RUN_TIME_STATS, CONFIG_FREERTOS_RUN_TIME_STATS_USING_ESP_TIMER.
I'm also fine with tutorial how to create the lib files with missing functions.

Thx

valeros, 2021-02-17:
Hi @SalsaMicha ! I'm not sure I understand your question. What framework do you use?

SalsaMicha, 2021-02-17:
Hi @valeros, I use vs code with platformio extentions and the ESP32 package. When I want to use the FreeRTOS Funktion uxTaskGetState() I get a comiple Error saying "undefined reference" even with all the defines mentioned above. After searching I found out, that the  libFreeRTOS.a within the ESP32 package does't contain the uxTaskGetState when it was builded. It has to be switched on in the SDKconfig when create the platformio(ardunio) package.

valeros, 2021-02-17:
It means that you need to open an issue in the https://github.com/espressif/arduino-esp32 repository as we don't compile that libraries we only use already their prebuilt package.

closed 2021-02-17 by valeros