platformio/platform-espressif32 #495: Build Error: Multiple ways to build the same target on platform-espressif32 v3.0.0

rg2k, 2021-02-21:
Hi,

I have the following errors when I update to platform-espressif32 v3.0.0 and while using arduino-esp32 v4.0.1. Works fine with platform-espressif32 v2.1.0.

Error 1:
```
warning: Two different environments were specified for target \.pio\build\esp32dev\port.o,
        but they appear to have the same action: ${TEMPFILE('$CC -o $TARGET -c $CFLAGS $CCFLAGS $_CCCOMCOM $SOURCES','$CCCOMSTR')}
File "C:\PlatformIO\storage\40\platforms\espressif32\builder\frameworks\espidf.py", line 609, in compile_source_files

*** Multiple ways to build the same target were specified for: \.pio\build\esp32dev\port.o  (from ['C:\\PlatformIO\\storage\\40\\packages\\framework-espidf\\components\\freemodbus\\port\\port.c'] and from ['C:\\PlatformIO\\storage\\40\\packages\\framework-espidf\\components\\freertos\\port.c'])
File "C:\PlatformIO\storage\40\platforms\espressif32\builder\frameworks\espidf.py", line 609, in compile_source_files
```

Error 2:
```
*** Multiple ways to build the same target were specified for: \.pio\build\esp32dev\sntp.o  (from ['C:\\PlatformIO\\storage\\40\\packages\\framework-espidf\\components\\lwip\\apps\\sntp\\sntp.c'] and from ['C:\\PlatformIO\\storage\\40\\packages\\framework-espidf\\components\\lwip\\lwip\\src\\apps\\sntp\\sntp.c'])
File "C:\PlatformIO\storage\40\platforms\espressif32\builder\frameworks\espidf.py", line 609, in compile_source_files
```

platformio.ini:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino, espidf

[platformio]
core_dir = c:/PlatformIO/storage/40
```

valeros, 2021-02-22:
Hi @rg2k ! Could you please put together a simple example to reproduce the issue?

rg2k, 2021-02-23:
Hi @valeros,

It seems to be an issue mixing 2.10 and 3.0.0 projects in the same storage folder. Separating the folder in the projects platformio.ini appears to have resolved the issue.

closed 2021-02-23 by rg2k