platformio/platform-espressif32 #497: ESP32-C3 Support

Robertou2, 2021-02-22:
Support for ESP32-C3 would be appreciated.

Jason2866, 2021-04-14:
Initial Arduino support is out for C3. To use the git version the gcc riscv toolchains would be needed.

Jason2866, 2021-04-20:
@Robertou2  i have done the support for the C3 with a custom setup.
You can use this settings to get it going:
```
; *** EXPERIMENTAL Tasmota version for ESP32-C3
[env:tasmota32c3]
extends                     = env:tasmota32_base
board                       = esp32c3
platform                    = https://github.com/Jason2866/platform-espressif32.git#feature/arduino-c3
platform_packages           = framework-arduinoespressif32 @ https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/v.2.0.0.pre/framework-arduinoespressif32-master-cf457d412.tar.gz
                              tasmota/toolchain-riscv32
````
[board](https://github.com/arendst/Tasmota/blob/development/boards/esp32c3.json) is defined here and has to placed in folder `boards` in root folder of your project

Robertou2, 2021-04-20:
@Jason2866 Thanks for your information. Following your information I have run Tflite Hello world in an ESP32C3 using the Arduino IDE
Now I am using your configuration but I get this error

Processing tasmota32c3 (board: esp32c3; platform: https://github.com/Jason2866/platform-espressif32.git#feature/arduino-c3)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32c3.html
PLATFORM: Espressif 32 (3.0.0+sha.71f92b5) > Espressif ESP32-C3
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 0.0.0
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-riscv32 1.80400.210415 (8.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 3 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\tasmota32c3\src\main.o
**riscv32-esp-elf-g++: error: unrecognized command line option '-mlongcalls'**
*** [.pio\build\tasmota32c3\src\main.o] Error 1
============================================================================================ [FAILED] Took 2.05 seconds ============================================================================================
The terminal process "C:\Users\roberto\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

I using Platformio in VSCode in windows 10

Thanks

Jason2866, 2021-04-20:
Just tested and it does work.
```
> Executing task: C:\Python37\Scripts\platformio.exe run <

Processing tasmota32c3 (board: esp32c3; platform: https://github.com/Jason2866/platform-espressif32.git#feature/arduino-c3; framework: arduino)
--------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32c3.html
PLATFORM: Espressif 32 (3.0.0+sha.71f92b5) > Espressif ESP32-C3
HARDWARE: ESP32C3 80MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 0.0.0
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-mklittlefs 1.203.210203 (2.3)
 - toolchain-riscv32 1.80400.210415 (8.4.0)
Converting tasmota.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP32-to-ESP8266-compat> 0.0.2
|   |-- <WiFi> 1.0
|   |-- <HTTPClient> 1.2
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <HTTPUpdate> 1.3
|   |   |-- <HTTPClient> 1.2
|   |   |   |-- <WiFi> 1.0
|   |   |   |-- <WiFiClientSecure> 1.0
|   |   |   |   |-- <WiFi> 1.0
|   |   |-- <Update> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <WebServer> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <FS> 1.0
|-- <WiFi> 1.0
|-- <Berry> 7.0
|   |-- <FS> 1.0
|-- <EEPROM 24C128_256_521>
|   |-- <Wire> 1.0.1
|-- <FFat> 1.0
|   |-- <FS> 1.0
|-- <FS> 1.0
|-- <UdpListener> 1.0
|-- <Unishox Compressor Decompressor highly customized and optimized for ESP8266 and Tasmota> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <M5 Stack Core2 library> 1.0
|   |-- <Wire> 1.0.1
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <headers>
|   |-- <EEPROM> 1.0.3
|   |-- <ESP32-to-ESP8266-compat> 0.0.2
|   |   |-- <WiFi> 1.0
|   |   |-- <HTTPClient> 1.2
|   |   |   |-- <WiFi> 1.0
|   |   |   |-- <WiFiClientSecure> 1.0
|   |   |   |   |-- <WiFi> 1.0
|   |   |-- <HTTPUpdate> 1.3
|   |   |   |-- <HTTPClient> 1.2
|   |   |   |   |-- <WiFi> 1.0
|   |   |   |   |-- <WiFiClientSecure> 1.0
|   |   |   |   |   |-- <WiFi> 1.0
|   |   |   |-- <Update> 1.0
|   |   |   |-- <WiFi> 1.0
|   |   |-- <ESPmDNS> 1.0
|   |   |   |-- <WiFi> 1.0
|   |   |-- <WebServer> 1.0
|   |   |   |-- <WiFi> 1.0
|   |   |   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <Ext-printf> 1.0
|-- <JSMN JSON parser customized and optimized for ESP8266 and Tasmota> 1.0
|-- <LittleFS_esp32> 1.0.5
|   |-- <FS> 1.0
|-- <PubSubClient> 2.8
|-- <SD(esp32)> 1.0.5
|   |-- <FS> 1.0
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <TasmotaSerial> 3.3.0
|   |-- <ESP32-to-ESP8266-compat> 0.0.2
|   |   |-- <WiFi> 1.0
|   |   |-- <HTTPClient> 1.2
|   |   |   |-- <WiFi> 1.0
|   |   |   |-- <WiFiClientSecure> 1.0
|   |   |   |   |-- <WiFi> 1.0
|   |   |-- <HTTPUpdate> 1.3
|   |   |   |-- <HTTPClient> 1.2
|   |   |   |   |-- <WiFi> 1.0
|   |   |   |   |-- <WiFiClientSecure> 1.0
|   |   |   |   |   |-- <WiFi> 1.0
|   |   |   |-- <Update> 1.0
|   |   |   |-- <WiFi> 1.0
|   |   |-- <ESPmDNS> 1.0
|   |   |   |-- <WiFi> 1.0
|   |   |-- <WebServer> 1.0
|   |   |   |-- <WiFi> 1.0
|   |   |   |-- <FS> 1.0
|-- <Ticker> 1.1
|-- <WiFiClientSecure> 1.0
|   |-- <WiFi> 1.0
|-- <Wire> 1.0.1
Building in release mode
*** use provided user_config_override.h as planned ***
*** use provided platformio_override.ini as planned ***
Use manually specified: COM4
Retrieved `.pio\build\tasmota32c3\src\StackThunk_light.cpp.o' from cache
Compiling .pio\build\tasmota32c3\src\WiFiClientSecureLightBearSSL.cpp.o
Compiling .pio\build\tasmota32c3\src\homekit.c.o
Compiling .pio\build\tasmota32c3\src\tasmota.ino.cpp.o
Retrieved `.pio\build\tasmota32c3\partitions.bin' from cache
```
Delete hidden folder `.cache` and `.platformio` forcing to download the latest versions of the packages.
Forgot to mention that you need a custom boards definition. Use [this](https://github.com/arendst/Tasmota/blob/development/boards/esp32c3.json)  and placed it in folder `boards` in root of your project.
Reason at the moment there is no definition for C3 boards for Arduino (only for IDF)

Robertou2, 2021-04-20:
Thank you, working fine. You only  have to add  framework  = arduino to the config file.

Jason2866, 2021-04-20:
Happy that it does work for you too.

Robertou2, 2021-04-21:
@Jason2866 Have you run Tasmota in the ESP32-C3?

Jason2866, 2021-04-21:
> @Jason2866 Have you run Tasmota in the ESP32-C3?

Yes, in a very early version. Wifimanager not working. Credentials hardcoded and some other (small)
quirks solved quick and dirty.

thijstriemstra, 2021-04-25:
@Jason2866 I tried your config:

```
[env:test_env]
board = esp32c3
framework = arduino
platform = https://github.com/Jason2866/platform-espressif32.git#feature/arduino-c3
platform_packages = framework-arduinoespressif32 @ https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/v.2.0.0.pre/framework-arduinoespressif32-master-cf457d412.tar.gz
```

But when trying to use SoftwareSerial (using EspSoftwareSerial package) I'm running into this issue: https://github.com/espressif/esp-idf/issues/6463

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32c3.html
PLATFORM: Espressif 32 (3.0.0+sha.71f92b5) > Espressif ESP32-C3
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 0.0.0 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-riscv32 1.80400.210415 (8.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 92 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <EspSoftwareSerial> 6.12.2
|-- <MD_YX5300> 1.3.1
|-- <YX5300_AudioPlayer>
|   |-- <MD_YX5300> 1.3.1
|   |-- <EspSoftwareSerial> 6.12.2
Building in release mode
Compiling .pio/build/xy5300_mp3/src/main.cpp.o
Linking .pio/build/xy5300_mp3/firmware.elf
/home/foo/.platformio/packages/toolchain-riscv32/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: .pio/build/xy5300_mp3/lib53d/libEspSoftwareSerial.a(SoftwareSerial.cpp.o): in function `std::__atomic_base<bool>::exchange(bool, std::memory_order)':
/home/foo/.platformio/packages/toolchain-riscv32/riscv32-esp-elf/include/c++/8.4.0/bits/atomic_base.h:413: undefined reference to `__atomic_exchange_1'
collect2: error: ld returned 1 exit status
*** [.pio/build/xy5300_mp3/firmware.elf] Error 1
```

Any idea how to fix this?

Jason2866, 2021-04-26:
Keep in mind Arduino support for C3 is in early alpha state. I just did a setup to get it running with platformio. Any issues related to Arduino ESP32 should be opened there https://github.com/espressif/arduino-esp32

KMeldgaard, 2021-05-27:
Will ESP32-C3 support for ESP-IDF (and ESP-IDF/Arduino hybrid) also be supported in this feature request? @valeros?

Leytronix, 2021-06-22:
Hi,

Am getting an error that no compatible libraries can be found (see below). How to solve this?

Thanx,

Tjerk

Processing tasmota32c3 (board: esp32c3; platform: https://github.com/Jason2866/platform-espressif32.git#feature/arduino-c3)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32c3.html
PLATFORM: Espressif 32 (3.2.0+sha.e01f799) > Espressif Generic ESP32-C3 4M Flash, Tasmota 1856k Code/OTA, 320k SPIFFS
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)      
PACKAGES:
 - framework-arduinoespressif32 0.0.0
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-riscv32 1.80400.210415 (8.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\tasmota32c3\src\main.o
riscv32-esp-elf-g++: error: unrecognized command line option '-mlongcalls'
*** [.pio\build\tasmota32c3\src\main.o] Error 1
========================================================================= [FAILED] Took 1.07 seconds =========================================================================The terminal process "C:\Users\Tjerk\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Jason2866, 2021-06-22:
@Leytronix  You are not using the latest Tasmota development version.
Just tried Tasmota compiles successfully for the ESP32-C3
Tasmota is build via LDF option chain and strict. You have chain and soft.
So something is different in the setup you use.

Leytronix, 2021-06-22:
Hi Jason,

Thx for your reply but strange. I just used the settings you proposed here above:

[env:test_env]
board = esp32c3
framework = arduino
platform   = https://github.com/Jason2866/platform-espressif32.git#feature/arduino-c3
platform_packages   = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
                                     tasmota/toolchain-riscv32

Jason2866, 2021-06-22:
And this is NOT a setting which will work to compile Tasmota. Tasmota has a complex Platformio setup. Just git clone Tasmota. Do run one build of normal Tasmota ` pio run -e tasmota`
after doing this special build variants are activated. You can build the c3 variant no with
`pio run -e tasmot32c3`
The C3 [env] is
```
; *** EXPERIMENTAL Tasmota version for ESP32-C3
[env:tasmota32c3]
extends                     = env:tasmota32_base
board                       = esp32c3
platform                    = https://github.com/Jason2866/platform-espressif32.git#feature/arduino-c3
platform_packages           = framework-arduinoespressif32 @ https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/307/framework-arduinoespressif32-master-1d7068e4b.tar.gz
                              tasmota/toolchain-riscv32
                              platformio/tool-mklittlefs @ ~1.203.200522
build_unflags               = ${esp32_defaults.build_unflags}
                              -Wswitch-unreachable
                              -mtarget-align
build_flags                 = ${esp32_defaults.build_flags}
                              -Wno-switch-unreachable
                              ;-DESP32_STAGE=true
lib_extra_dirs              = lib/libesp32
                              lib/libesp32_lvgl
                              lib/lib_basic
                              lib/lib_i2c
                              lib/lib_ssl
                              lib/lib_display
lib_ignore                  =
                              NimBLE-Arduino
                              Micro-RTSP
```
which will NOT work standalone to build Tasmota! There are flags and settings in other sections too

Leytronix, 2021-06-22:
Thanks for the information but unfortunately this sounds a bit complicated to me. When do you think a less experimental version will be available which will work without having to build it first?

Jason2866, 2021-06-22:
You can find special Tasmota automated builds (every hour) here https://github.com/Jason2866/Tasmota-specials
I added the experimental Tasmota32-C3 build.

Leytronix, 2021-06-23:
I now have cloned the Tasmota-specials and want to flash the C3 binary to my C3 board connected to COM6 but something goes wrong with a path (see below). From what folder do I need to start the command?

Thanks,

Tjerk

--------------------------------------------------------------------------------------------------------------------------------------------
C:\Users\Tjerk\Desktop\Tasmota-specials-firmware>esptool.py --chip esp32 --port COM6 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dout --flash_freq 40m --flash_size detect 0x1000 bootloader_dout_40m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 tasmota32c3.bin
usage: esptool write_flash [-h] [--erase-all] [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}] [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress] [--verify] [--encrypt]
                           [--ignore-flash-encryption-efuse-setting] [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'bootloader_dout_40m.bin'

Jason2866, 2021-06-23:
**Ahh forgot!** I do flashing via VSC so i dont have to take care for the right bootloader and addresses. C3 has a different bootloader and flashing addresses are different!
Hold on for now. I have to add the C3 bootloader, partition.bin and the flash command string.
Will have time for in a few hours.

Jason2866, 2021-06-23:
Added the C3 needed files and esptool.py command.
It is here: https://github.com/Jason2866/Tasmota-specials/tree/firmware/firmware/tasmota32/ESP32-C3_needed_files

Jason2866, 2021-06-23:
For further questions to Tasmota please use our Discussions section. https://github.com/arendst/Tasmota/discussions We hijacked this issue.

thijstriemstra, 2021-06-23:
> We hijacked this issue.

Agreed.

> When do you think a less experimental version will be available which will work without having to build it first?

I have the same question.

Jason2866, 2021-06-23:
Theo Arendst, do not release a binary which is not verified to run without a WDT for a long time with the default features working. There are many parts in Arduino core (IDF?) still not working / ported. So thats why "experimental". Next problem. You can hardly buy a ESP32-C3. Theo has none. The port for the C3 was done from Stephan. He is the contributer who wrote all zigbee stuff for Tasmota. Anyway you find in my repo a unofficial build from latest development for the C3.
https://github.com/Jason2866/Tasmota-specials

closed 2021-06-23 by valeros

thijstriemstra, 2021-06-30:
> When do you think a less experimental version will be available which will work without having to build it first?

[v3.3.0](https://github.com/platformio/platform-espressif32/releases/tag/v3.3.0) added support!

Jason2866, 2021-06-30:
@thijstriemstra  Yes, now the alpha build is supported to build with platformio. So my custom setup to build the C3 version is not needed anymore. The resulting Tasmota version is still experimental

Tinyu-Zhao, 2021-11-14:
> boards

I configured the program as you described and compiled the program and prompted me "riscv32-esp-elf-g++: error: unrecognized command line option'-mlongcalls'"

`
> Executing task in folder STAMP-C3: platformio run <

Processing tasmota32c3 (board: esp32c3; platform: https://github.com/Jason2866/platform-espressif32.git#feature/arduino-c3)
---------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32c3.html
PLATFORM: Espressif 32 (3.2.0+sha.e01f799) > Espressif Generic ESP32-C3 4M Flash, Tasmota 1856k Code/OTA, 320k SPIFFS
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 0.0.0 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - toolchain-riscv32 1.80400.210415 (8.4.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/tasmota32c3/src/main.o
riscv32-esp-elf-g++: error: unrecognized command line option '-mlongcalls'
*** [.pio/build/tasmota32c3/src/main.o] Error 1
=================================== [FAILED] Took 0.54 seconds ===================================

`

Jason2866, 2021-11-14:
@Tinyu-Zhao you have not cloned the whole actual repo. There are build_flags missing.
The platformio setup is done not just in one file.

JamesSmartCell, 2021-11-29:
> @Tinyu-Zhao you have not cloned the whole actual repo. There are build_flags missing. The platformio setup is done not just in one file.
> 
> ```
> PLATFORM: Espressif 32 (3.3.1+sha.33ff413) > Espressif Generic ESP32-C3 4M Flash, Tasmota 1856k Code/OTA, 320k SPIFFS
> HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
> DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
> PACKAGES: 
>  - framework-arduinoespressif32 2.0.1 
>  - tool-esptoolpy 1.30200.211025 (3.2.0) 
>  - tool-mklittlefs 1.203.210628 (2.3) 
>  - toolchain-riscv32-esp 8.4.0+2021r1
> Converting tasmota.ino
> LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ strict
> Found 118 compatible libraries
> ```
> 
> ```
> Linking .pio/build/tasmota32c3/firmware.elf
> Retrieving maximum program size .pio/build/tasmota32c3/firmware.elf
> Checking size .pio/build/tasmota32c3/firmware.elf
> Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
> RAM:   [==        ]  16.2% (used 53032 bytes from 327680 bytes)
> Flash: [=======   ]  71.1% (used 1352092 bytes from 1900544 bytes)
> Building .pio/build/tasmota32c3/firmware.bin
> esptool.py v3.2
> Merged 2 ELF sections
> bin_map_copy([".pio/build/tasmota32c3/firmware.bin"], [".pio/build/tasmota32c3/firmware.elf"])
> map_gzip([".pio/build/tasmota32c3/firmware.bin"], [".pio/build/tasmota32c3/firmware.elf"])
> ======================== [SUCCESS] Took 222.34 seconds ========================
> 
> Environment    Status    Duration
> -------------  --------  ------------
> tasmota32c3    SUCCESS   00:03:42.343
> ========================= 1 succeeded in 00:03:42.343 =============
> ```

Where do you clone the repo to? I'm also stuck on the same error (-mlongcalls directive to assembler), and not sure where to clone the repo, or what to do after cloning.

Sorry for being a noob - it looks straightforward once you know what's going on. I did try to read the Tasmota install instructions but they just said the same thing - clone the repo - but didn't say what to do afterwards to setup in PlatformIO. I also tried running the ```pio run -e tasmota``` but that didn't seem to go anywhere.

The NimBLE library is listed under the 'ignore' section ... does it mean that this library can't be used, or that the library doesn't need to be converted?

Just got my ESP32C3's, looks like a perfect solution for most of my builds which don't need the dual core; also it looks like we can also resurrect all our old ESP8266 circuit board designs :)

Thanks :)

Jason2866, 2021-11-30:
@JamesSmartCell  It does work for Arduino. Working setup https://github.com/platformio/platform-espressif32/issues/619#issuecomment-976525911
NimBLE (h2zero version) is working since a while with the C3 too
We run Tasmota with BLE functions on the C3.

JamesSmartCell, 2021-12-01:
> @JamesSmartCell It does work for Arduino. Working setup [#619 (comment)](https://github.com/platformio/platform-espressif32/issues/619#issuecomment-976525911) NimBLE (h2zero version) is working since a while with the C3 too We run Tasmota with BLE functions on the C3. For asking Tasmota build & compile question visit the [Tasmota Discord](https://discord.gg/Ks2Kzd4)

```
[env:esp32dev]
platform                    = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master
platform_packages           = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.1
                              platformio/tool-esptoolpy @ https://github.com/tasmota/esptool/releases/download/v3.2/esptool-v3.2.zip
```

Thanks, it looks like I'm getting closer, however, I still couldn't get it working as none of the board settings worked:

```
[env:esp32dev]
platform                    = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master
framework = arduino
board = esp32-c3
platform_packages           = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.1
                              platformio/tool-esptoolpy @ https://github.com/tasmota/esptool/releases/download/v3.2/esptool-v3.2.zip
monitor_speed = 115200
```

This fails as there's no esp32-c3 board definition. I used the esp32-c3-devkit.json definition from the working ESPIDF config, but this produces a lot of compile errors.

I pinged the forum as I'm probably overlooking some fundamental point that's too obvious!

Jason2866, 2021-12-01:
@JamesSmartCell 
```
board = esp32-c3-devkitm-1
```

JamesSmartCell, 2021-12-01:
> @JamesSmartCell
> 
> ```
> board = esp32-c3-devkitm-1
> ```

I tried that one too, it crunches through a few library files but then errors on Stream.cpp

```
Compiling .pio\build\esp32dev\FrameworkArduino\Stream.cpp.o
In file included from /.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/freertos/port/riscv/include/freertos/portmacro.h:38,
                 from /.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/freertos/include/freertos/portable.h:51,
                 from /.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/freertos/include/freertos/FreeRTOS.h:63,
                 from /.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:33,
                 from src/main.cpp:1:
/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/esp_hw_support/include/soc/spinlock.h: In function 'bool spinlock_acquire(spinlock_t*, int32_t)': 
/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/esp_hw_support/include/soc/spinlock.h:67:36: error: 'XCHAL_EXCM_LEVEL' was not declared in this scope
     irq_status = XTOS_SET_INTLEVEL(XCHAL_EXCM_LEVEL);
                                    ^~~~~~~~~~~~~~~~
```

JamesSmartCell, 2021-12-01:
> @JamesSmartCell
> 
> ```
> board = esp32-c3-devkitm-1
> ```

I finally got it 100% working. Using your suggestions to get to the 'conflicting priv spec version' error, I was able to find this solution:

https://community.platformio.org/t/broken-esp32-c3-config-build-after-moving-to-arduino-esp32-v2-0-0-rc1/22817

Here's my full, working platform.io for anyone hitting this issue (at least until official esp32c3 Arduino support is out):

`
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
platform_packages =
    toolchain-riscv-esp
	framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.1
	platformio/tool-esptoolpy @ https://github.com/tasmota/esptool/releases/download/v3.2/esptool-v3.2.zip
framework = arduino
board = esp32dev
board_build.mcu = esp32c3
board_build.partitions = huge_app.csv
board_build.variant = esp32c3
board_build.f_cpu = 160000000L
board_build.f_flash = 80000000L
board_build.flash_mode = dio
board_build.arduino.ldscript = esp32c3_out.ld
build_unflags =
	-DARDUINO_ESP32_DEV
	-DARDUINO_VARIANT="esp32"
build_flags =
	-DARDUINO_ESP32C3_DEV
	-DARDUINO_VARIANT="esp32c3"
    -DARDUINO_USB_CDC_ON_BOOT=0
build_type = debug
monitor_speed = 115200
`

It does give a strange error when you boot your flash:

`
SHA-256 comparison failed:
Calculated: 071318b1de86cde7f1aec94a59e4cb69b9476060b25c8688588480d5e8f36d1c
Expected: aeb43c2bf68a09f09f2b1d7c8bb79c7727cf0a2b827a76540c98a24fcafdd49a
Attempting to boot anyway...
`

This doesn't seem to affect operation, but I wouldn't want to deploy anything in the field with that.

Thank you @Jason2866 very much for your patience, I wouldn't have been able to get there without your help.

hpsaturn, 2021-12-27:
Hi, the bad thing with this workaround @JamesSmartCell it's that many libraries don't have this flag validation. The flag `esp32` that we did `build_unflag`, it is used on the validation into esp32 vs esp8266. We will need do many pull requests to have again these libraries.

Jason2866, 2021-12-27:
@hpsaturn  Doing like @JamesSmartCell is not the recommended nor necessary.
He added definitions which are not needed.
NEVER redefine a boards.json!! 
Make [env] you want with the needed board. Thats all!
Nowhere is the definition of the toolchain needed!

commit b073ed84 (2021-06-23): Initial support for ESP32C3

Resolve #568, resolve #497