platformio/platform-espressif32 #498: platform = url

IZ76, 2021-02-24:
platform = https://github.com/platformio/platform-espressif32.git
эта фишка не работает.
Processing mhetesp32devkit (platform: https://github.com/platformio/platform-espressif32.git; board: esp32doit-devkit-v1; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing git+https://github.com/platformio/platform-espressif32.git
Error: Traceback (most recent call last):
  File "c:\users\ivan\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 70, in process
    p = PlatformFactory.new(self.options["platform"])
и так далее на целую страницу

valeros, 2021-02-24:
Hi @IZ76 ! Please attach here the entire log so we can gain a better understanding of the problem.

IZ76, 2021-02-25:
> Executing task in folder VZ_MEDIA_CENTER_V_1.0: C:\Users\ivan_admin\.platformio\penv\Scripts\pio.exe run <

Processing mhetesp32devkit (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing git+https://github.com/platformio/platform-espressif32.git
Error: Traceback (most recent call last):
  File "c:\users\ivan_admin\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "c:\users\ivan_admin\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\ivan_admin\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\ivan_admin\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\ivan_admin\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\ivan_admin\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\ivan_admin\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\ivan_admin\.platformio\penv\lib\site-packages\click\decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\ivan_admin\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 143, in cli
    is_test_running,
  File "c:\users\ivan_admin\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 175, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "c:\users\ivan_admin\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 79, in process
    result = p.run(build_vars, build_targets, self.silent, self.verbose, self.jobs)
  File "c:\users\ivan_admin\.platformio\penv\lib\site-packages\platformio\platform\_run.py", line 60, in run
    self.autoinstall_runtime_packages()
  File "c:\users\ivan_admin\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 74, in autoinstall_runtime_packages
    self.pm.install(self.get_package_spec(name))
  File "c:\users\ivan_admin\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 49, in install
    spec, silent=silent, skip_dependencies=skip_dependencies, force=force
  File "c:\users\ivan_admin\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97, in _install
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "c:\users\ivan_admin\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 141, in install_from_url
    vcs = VCSClientFactory.new(tmp_dir, url)
  File "c:\users\ivan_admin\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 55, in new
    src_dir, remote_url, tag, silent
  File "c:\users\ivan_admin\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 137, in __init__
    self.configure()
  File "c:\users\ivan_admin\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 146, in configure
    result = proc.exec_command([cls.command, "--exec-path"])
  File "c:\users\ivan_admin\.platformio\penv\lib\site-packages\platformio\proc.py", line 117, in exec_command
    p = subprocess.Popen(*args, **kwargs)
  File "C:\Users\ivan_admin\.platformio\python3\lib\subprocess.py", line 800, in __init__
    restore_signals, start_new_session)
  File "C:\Users\ivan_admin\.platformio\python3\lib\subprocess.py", line 1207, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] Не удается найти указанный файл

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

Процесс терминала "C:\Users\ivan_admin\.platformio\penv\Scripts\pio.exe 'run'" был завершен с кодом выхода 1.

PS. Использую пример из этой библиотеки.
https://github.com/schreibfaul1/ESP32-audioI2S
На АрдуиноИДЕ удалось запустить данную библиотеку после обновления платы ESP32 Arduino до version 1.0.5

valeros, 2021-02-25:
You need to install Git beforehand https://git-scm.com/book/en/v2/Getting-Started-Installing-Git

closed 2021-02-25 by valeros

IZ76, 2021-02-25:
Спасибо все работает.