platformio/platform-espressif32 #503: Feature request: LDF scan for a new entry "mcu"

Jason2866, 2021-02-27:
- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

### Configuration

**Operating system**: All

**PlatformIO Version** (`platformio --version`): latest development 

### Description of problem
Add a option to the LDF to scan for `mcu` in librarary.properties or in library.json
`mcu` is already a entry in Arduino ESP32  [boards.txt](https://github.com/espressif/arduino-esp32/blob/idf-release/v4.2/boards.txt#L30)
Esp32 has different mcu cores now (ESP32-C3 soon too) which needs different libraries.
At the moment there is no possibility to differentiate this or did i missed something?
Maybe as new command `strict+` ?

ivankravets, 2021-02-27:
It should be implemented in dev-platform. See https://github.com/espressif/arduino-esp32/blob/master/tools/platformio-build.py#L177

valeros, 2021-03-01:
HI @Jason2866 ! Could you please provide an example of any library that needs this feature?

Jason2866, 2021-04-20:
The ESPP3232-C3 is available. All libs are used which are compatible to `core esp32` for the C3 as it is now since
there is no possibility to handle the mcu variant with LDF.
We have some optimized libs which are mcu dependent ( xtensa vs. riscv) it would be nice not to litter the code 
with many `#ifdefs` to take care of this. More elegant would be to have lib variants for every architecture