platformio/platform-espressif32 #506: xtensa-esp32s2-elf-gcc: error: unrecognized command line option '-Wdocumentation'

markus-seidl, 2021-03-01:
Hi!

I have the following CMakeLists.txt

```
cmake_minimum_required(VERSION 3.16.0)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(esp32s2-ulp-test)
```

and this platformio.ini

```

[env:default]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32doit-devkit-v1
framework = espidf
board_build.mcu = esp32s2

```

Unfortunately ```pio run``` gives me:

```
Compiling .pio/build/default/mbedtls/mbedtls/library/aesni.o
xtensa-esp32s2-elf-gcc: error: unrecognized command line option '-Wdocumentation'
```

I tried to remove these errors by not compiling mbedtls, but this doesn't seem possible (it's just used too much...)

I was mix and matching cmake configurations with another project, that includes the arduino platform (which works for esp32s2), unfortunately I'm neither fluent nor good in the IDF to spot the error here.

I've also tried

```
cmake_minimum_required(VERSION 3.16.0)

set(CMAKE_SYSTEM_NAME Generic)
set(CMAKE_C_COMPILER_WORKS 1)
set(CMAKE_CXX_COMPILER_WORKS 1)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(esp32s2-ulp-test  C CXX)

SET(CMAKE_C_COMPILER "$ENV{HOME}/.platformio/packages/toolchain-xtensa32s2/bin/xtensa-esp32s2-elf-gcc")
SET(CMAKE_CXX_COMPILER "$ENV{HOME}/.platformio/packages/toolchain-xtensa32s2/bin/xtensa-esp32s2-elf-g++")
SET(CMAKE_CXX_FLAGS "-mlongcalls -std=gnu++11 -fexceptions -fno-rtti -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -ggdb -O2 -fstack-protector -MMD")
SET(CMAKE_C_FLAGS "-mlongcalls -std=gnu99 -Wno-old-style-declaration -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -ggdb -O2 -fstack-protector -MMD")
```

But it seems I cannot remove the "-Wdocumentation" to my knowledge.

valeros, 2021-03-01:
Hi @markus-seidl ! Do you have Clang installed in your system? It seems that CMake picks it up for some reason, even though GCC should have a higher priority.

markus-seidl, 2021-03-01:
Yes 

```
clang --version
Apple clang version 12.0.0 (clang-1200.0.32.29)
Target: x86_64-apple-darwin20.3.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

I've created the project with Clion and got errors while creating.
I've just installed Visual Studio Code, created a new project and it now compiles in both projects.

valeros, 2021-03-01:
FYI, CLion cannot be used together with ESP-IDF and PlatformIO because of a clash between CMake files.

closed 2021-03-01 by valeros

markus-seidl, 2021-03-01:
Thanks for the prompt answer! I've came from the Arduino Expressif PlatfomIO platform, which didn't have any problems (or maybe I was just lucky).

valeros, 2021-03-01:
Only the ESP-IDF integration is affected, all other frameworks can be used with CLion.