platformio/platform-espressif32 #508: Update framework-arduinoespressif32 package with hotfixes

maxgerhardt, 2021-03-06:
After the release of the 1.0.5 Arduino-ESP32 core, a few important bugfixes have been made which also affect their usage in PlatformIO ([refer commit history](https://github.com/espressif/arduino-esp32/commits/master)). 

1. Creating a project with `board = firebeetle32` results in total compilation failure because of an error in the C++ code regarding duplicate pin definitions.  [Reported by user here](https://community.platformio.org/t/firebeetle-compiling/19758) and fixed in https://github.com/espressif/arduino-esp32/commit/6e7cc5210dd210603f4e49cf677ccfa8e055b531
2. 3 seperate users have reported seeing warnings stemming from the `c_cpp_properties.json` refernecing a non-existing folder, `tools/sdk/include/nimbus`, in the Arduino core ([See reports](https://community.platformio.org/t/esp32-3-1-0-missing-files/19696)).  Fixed by my PR 2 days ago in [here](https://github.com/espressif/arduino-esp32/pull/4891). 

Thus I think it makes sense to repackage the Arduino-ESP32 core with these bugfixes -- I'm not sure if the Arduino core will release a 1.0.6 version for that or if it's longer.

valeros, 2021-03-08:
Thanks for reporting, fixed directly in the framework package.

maxgerhardt, 2021-03-09:
@valeros, a performance degradation was noted in https://community.platformio.org/t/very-bad-performance-if-i-compile-the-example-for-esp32-cam/19730/10?u=maxgerhardt 

>When I compile the example in Arduino I get around 15 fps from the camera.
If I import it into Platformio I get 0.2 fps

for which the PR https://github.com/espressif/arduino-esp32/pull/4911 is a fix. So I think a re-release is noted again. 

**But** it's also reported that other basic functionality like `String::replace()` is broken (https://github.com/espressif/arduino-esp32/issues/4920) so it seems like a rollercoast ride right now. Maybe you can just update the package with the PR from above to fix the performance regression.

maxgerhardt, 2021-03-10:
Then maybe the user had an older version of the framework-arduinoespressif32 package that caused this, hm.

me-no-dev, 2021-03-10:
@maxgerhardt the new version 1.0.6 will be released soon and will include the fixes. @valeros do I need to let you know?

valeros, 2021-03-10:
@me-no-dev Would be great.

me-no-dev, 2021-03-26:
@valeros 1.0.6 is out :)

commit 751be0cd (2021-03-26): Update Arduino package to v1.0.6 // Resolve #508
closed 2021-03-26 by valeros