platformio/platform-espressif32 #512: Permission denied error for AsyncTCP.a

AWhiteKnight, 2021-03-11:
Hello,

I am using platformio within visual code.
Since the last full update (platforms, packages, libraries),  I always get the following error for esp32 projects:
Archiving .pio\build\esp32Chassis\libd88\libAsyncTCP.a
xtensa-esp32-elf-ar: unable to rename '.pio\build\...\libAsyncTCP.a'; reason: Permission denied
*** [.pio\build\esp32Chassis\libd88\libAsyncTCP.a] Error 1

If I look into the directory, there is no libAsyncTCP.a

It is only for esp32
It is only this library, never any other library
It is in any esp32 compile that uses this library
It does not happen with the esp8266 version ESPAsyncTCP
It is with both
  ottowinter/AsyncTCP-esphome
  me-no-dev/AsyncTCP
It happens also with completly new created projects without any logic or other libs if only that lib is added

Workaround is to run Visual Code as Administrator
Once the file is created you can build without being an Administrator until that file needs to be compiled again.

Thanks for your great work on platformio.

Stephan

valeros, 2021-03-11:
Hi Stephan!  Strange but I cannot reproduce the issue. Is this behavior intermittent or it occurs each time when you build your project?

AWhiteKnight, 2021-03-11:
Hi valeros,

it happens whenever the lib needs a recompile, for example after a clean or a change of compile flags etc.
Then I need to do one build as Administrator ( I forgot to mention I am on Windows 10, but maybe you guessed that)

Stephan

AWhiteKnight, 2021-03-11:
I did some tests with esp32 builds. 
It seems that this problem occurs, when a library uses other libs and during build there is created a subdirectory like .../esp32/lib<xxx>/AsyncTCP containing all the .o files.
AsyncTCP seems to be always the first of these libraries in my projects to be Archived, all libraries Archived before do not use a subdirectory in lib<xxx>
If I press build several times in sequence without changing anything, other libs with a subdir in lib<xxx> like WiFi also show up with that issue

Hope that helps
Stephan

valeros, 2021-03-11:
Could you please put together a simple example to reproduce the issue?

AWhiteKnight, 2021-03-11:
created a new project "test" with platformio-wizard
main.cpp as generated
[test.zip](https://github.com/platformio/platform-espressif32/files/6124186/test.zip)
platformio.ini added lib:deps manually to 

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps =
	AsyncTCP

AWhiteKnight, 2021-03-11:
[test.zip](https://github.com/platformio/platform-espressif32/files/6124201/test.zip)

ivankravets, 2021-03-11:
Do you have 3rd party antivirus tools in a system?

AWhiteKnight, 2021-03-11:
No, only windows securitiy

AWhiteKnight, 2021-03-11:
Well, to me it looks like that message is wrong, saying "Permission denied" - the file most probably does not exist.
So maybe the real error occours when that libAsyncTCP.a should have been created

AWhiteKnight, 2021-03-12:
Error does not occur in a Linux environment

ivankravets, 2021-03-14:
I'm not sure that we can help here. I remember we had similar issue a few years ago. It was caused by OS and permission settings.

AWhiteKnight, 2021-03-16:
Why is it only for ESP32 not for ESP8266? It is a different behavior of the two toolchains within the same project.
And it is only for libs that need a subdirectory like lib839 and libabd in the following picture, it works for lib882 and libf33 which have no subdirectory
[image]
Starting the build process multiple times will proceed further in compiling the project.
All libs that create a subdirectory during build will fail with that error:

`Building in release mode
Compiling .pio\build\esp32\src\src\BotComponent.cpp.o
Compiling .pio\build\esp32\src\src\BotFeature.cpp.o
Compiling .pio\build\esp32\src\src\BotMesh.cpp.o
Compiling .pio\build\esp32\src\src\NodeBot.cpp.o
Compiling .pio\build\esp32\src\src\components\d1mini\D1Mini_Motor_Shield.cpp.o
Compiling .pio\build\esp32\src\src\main.cpp.o
Generating partitions .pio\build\esp32\partitions.bin
Archiving .pio\build\esp32\lib882\libTaskScheduler.a
Archiving .pio\build\esp32\libf33\libArduinoJson.a  
Compiling .pio\build\esp32\lib839\AsyncTCP\AsyncTCP.cpp.o
Compiling .pio\build\esp32\libabd\WiFi\ETH.cpp.o
Compiling .pio\build\esp32\libabd\WiFi\WiFi.cpp.o
Compiling .pio\build\esp32\libabd\WiFi\WiFiAP.cpp.o
Compiling .pio\build\esp32\libabd\WiFi\WiFiClient.cpp.o
Archiving .pio\build\esp32\lib839\libAsyncTCP.a
Compiling .pio\build\esp32\libabd\WiFi\WiFiGeneric.cpp.o
xtensa-esp32-elf-ar: unable to rename '.pio\build\esp32\lib839\libAsyncTCP.a'; reason: Permission denied
*** [.pio\build\esp32\lib839\libAsyncTCP.a] Error 1`

None of the files where Permission is denied exists, they are not created. Therefore the archiver cannot rename them.

Can you please give me a hint in which file at which line of the python scripts the *.a file should be created?
I will then try to find out what really happens on my machines.

Thanks
Stephan

AWhiteKnight, 2021-03-28:
I updated the arduino toolchain on my machine. Now I have the same issue with "permission denied" from ar.exe with arduinos.
A good guess is, that these errors occur due to a newer Microsoft Compiler used to create these newer releases of the tool-chains.
In both cases (esp32 and avr) its the ar - tool that has these problems. But not with the very old esp8266 toolchain.

closed 2022-09-20 by stale[bot]