platformio/platform-espressif32 #514: termux: ESP32 T-WATCH fails during "ar" tool (-31)

yesco, 2021-03-18:
I was hopeful to use pio on Termux ("unix for android") and indeed it installed and started downloading dependencies and crosscompilers etc for Espressif ESP32 embedded devices. Quite impressive.

In compiling https://github.com/sharandac/My-TTGO-Watch it fails during "ar" with an undocumented error -31. I'm unable to continue.

I've tried it on a remote VPS host (intel) and it compiled fine. So not sure what the issue is.

I've updated and upgraded pio and synced repo just now. It was clean.

Compiling .pio/build/ttgo-t-watch-v1/libb03/Ticker/Ticker.cpp.o               Archiving .pio/build/ttgo-t-watch-v1/libc50/libSPI.a                          *** [.pio/build/ttgo-t-watch-v1/libc50/libSPI.a] Error -31                    ==== [FAILED] Took 115.26 seconds ====
~/GIT/My-TTGO-Watch $

valeros, 2021-03-18:
Hi @yesco ! I've never seen that error before, so it's quite hard to guess what went wrong. Does it always fail on `libSPI.a`? Have you tried to compile any other project?

yesco, 2021-03-18:
I haven't found any other esp32 pio platform projects.

Pointers are welcome, and or a way to search for them specifically (in
github?)

Is there a mailing group for pio esp32 projects? description of libraries?

I'm at point where I'm trying to decide if it's going to be worth pursing
pio or not. I like command-line so it was promising, but when the espressif
executable is failing... It means it's not going to work on any ARM
archiecture? Host compiling on RPI maybe fails too?

valeros, 2021-03-18:
>  haven't found any other esp32 pio platform projects.

All examples are located here https://github.com/platformio/platform-espressif32/tree/develop/examples

> Is there a mailing group for pio esp32 projects? description of libraries? 

No, there is no such thing.

> It means it's not going to work on any ARM archiecture? Host compiling on RPI maybe fails too?

A lot of people use SBCs and PlatformIO to compile their projects, so I believe it should work without issues.

yesco, 2021-03-18:
Even blink fails compiling under Termux

I did a git clone and build of platform-espressif32/examples/arduino-blink

pio run, that is:

ameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/ttgo-lora32-v21/FrameworkArduino/IPAddress.cpp.o
*** [.pio/build/ttgo-lora32-v21/libFrameworkArduinoVariant.a] Error -31
Compiling .pio/build/ttgo-lora32-v21/FrameworkArduino/IPv6Address.cpp.o
===== [FAILED] Took 4.93 seconds =====
Environment          Status    Duration-------------------  --------
------------                                   esp wrover kit       FAILED
  00:07:45.678                                   esp32doit-devkit-v1
FAILED    00:00:05.477
lolin32              FAILED    00:00:05.092
   ttgo-lora32-v21      FAILED    00:00:04.929
      4 failed, 0 succeeded in 00:08:01.177

valeros, 2021-03-19:
I'm not sure how I can help you here, that `Error -31` is very strange. We can try to avoid creating static archives at all, but it doesn't seem worth the effort.

valeros, 2022-04-18:
Closing as no further info provided. Feel free to reopen the issue if you're still experiencing the problem with the latest upstream version of the platform.

closed 2022-04-18 by valeros