platformio/platform-espressif32 #515: [esp-idf] Failed build because “Multiple ways to build the same target were specified…”

Honza0297, 2021-03-19:
Hello!
When trying to build my project with esp-idf framework with PlatformIO IDE plugin in VSCode, the build fails as shown below:
~~~
> Executing task: platformio run --environment esp32dev <

Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40200.210118 (4.2.0) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - tool-ninja 1.7.1 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32 2.80400.210211 (8.4.0)
Reading CMake configuration...

*** Multiple ways to build the same target were specified for: /var/home/jaberan/Documents/PlatformIO/Projects/TOIcojavimkolikaty/.pio/build/esp32dev/esp_efuse_api.o  (from ['/var/home/jaberan/.platformio/packages/framework-espidf/components/efuse/src/esp32/esp_efuse_api.c'] and from ['/var/home/jaberan/.platformio/packages/framework-espidf/components/efuse/src/esp_efuse_api.c'])
File "/var/home/jaberan/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 607, in compile_source_files
==================================================================================== [FAILED] Took 3.22 seconds ====================================================================================
The terminal process "platformio 'run', '--environment', 'esp32dev'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
~~~

What I have tried:

  * General -> Clean
  * Reinstall esp-idf framework, platformio, even VSCode
  * Start new project with the same source code

Do somebody know what could be the culprit of that and how to solve that?  Thanks in advance!

valeros, 2021-03-19:
Hi @Honza0297 ! Could you please put together a minimal example to reproduce the issue?

Honza0297, 2021-03-19:
Certainly, however, during the preparation of the minimal example, I managed to build the project, with absolutely no change of anything. After I cleaned the project (General -> Clean), I am now unable to build it again. I'll dig deeper what changed (something had to) and provide you additional info.

Honza0297, 2021-03-19:
I somehow forgot to provide version info etc, may be helpful: 
OS: Fedora Silverblue
VSCode version 1.54.3 (flatpak) 
Platformio home: 3.3.4 
Platformio core: 5.1.1 
 
 Anyway, I found out what cause the error. My home directory (/home/jaberan) is a symlink to /var/home/jaberan. Sometimes, Platformio/cmake detects my $HOME prefix (or how to call that) as a /home (does not work), sometimes as a /var/home (works). The difference is present in CMakeFIles folder located in .pio/build/esp32dev. 
 Here is my project repository (simple program which gets temperature from ds18b20 temperature sensor and prints it to Serial line). Branch "broken" contains broken version, branch "working" contains working version. https://github.com/Honza0297/TOIcv1
 I am not sure if this information is somehow useful, but there is nothing more I could find.

Honza0297, 2021-03-22:
Update: After a little experimenting, I found out I can avoid the error by setting $HOME to "/var/home/jaberan" instead of "/home/jaberan". Then, I am able to create and compile working application. Not sure why PlatformIO (or esp-idf?) doesn't like my simlinked home...

chrissyan0, 2022-02-16:
This bug is also easily reproducable in a Windows portable environment:
- unpack current 64 bit zip version of vscode to d:\pTEST
```
> mkdir d:\pTEST\data\tmp
> set PLATFORMIO_CORE_DIR=d:\pTEST\data\extensions\.platformio
> code
```
- From the marketplace install platformio
- After restarting, in PIO home make a new project:
 -- Board AZ-Delivery ESP-32 Dev Kit  C V4
 -- Framework: Espressif IoT Dev. FW
 -- Custom project location somewhere on the D:

After having downloaded everything, the project wizard fails with that error:
```
*** Multiple ways to build the same target were specified for: D:\ESPFAIL\.pio\build\az-delivery-devkit-v4\esp_efuse_api.o  (from ['D:\\pTEST\\data\\extensions\\.platformio\\packages\\framework-espidf\\components\\efuse\\src\\esp32\\esp_efuse_api.c'] and from ['D:\\pTEST\\data\\extensions\\.platformio\\packages\\framework-espidf\\components\\efuse\\src\\esp_efuse_api.c'])  
File "D:\pTEST\data\extensions\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 620, in compile_source_files
```
Cleaning the project and rebuilding doesn't help. The same error rises after "Reading CMake configuration..."

However, it's working when you do the steps above omitting PLATFORMIO_CORE_DIR and let install everything in the user's homedirectory. Of course, on AD enabled machines with folder redirection, this is not an option.

VSCode:1.64.2
PIO: Core 5.2.5, Home: 3.4.1

Regards,
Chris

closed 2022-08-10 by stale[bot]

Ownezx, 2024-06-27:
I am having a similar issue as described here, my .platformio dir is a symlink to another OSX account. I cannot set my home directory to the symlink target as I don't want to affect other things.

AndrewCapon, 2024-10-25:
Same problem here with a symlinked `.platformio`

mwalser, 2025-01-06:
In case you are using the Flatpak version of VS Code on Fedora Silverblue or Kinoite, run the following command and restart VS Code:

```bash
flatpak override --user "--env=HOME=/var/home/$USER" com.visualstudio.code
```

This builds upon the discovery of @Honza0297 that PlatformIO apparently doesn't like symlinked home directories:

> Update: After a little experimenting, I found out I can avoid the error by setting $HOME to "/var/home/jaberan" instead of "/home/jaberan". Then, I am able to create and compile working application. Not sure why PlatformIO (or esp-idf?) doesn't like my simlinked home...