platformio/platform-espressif32 #516: SSL connection issues in (since?) platform version 3.1.0

maxgerhardt, 2021-03-22:
Per 3 user reports from https://community.platformio.org/t/wificlientsecure-doesnt-work-in-espressif-32-v3-1-0/19877, establishing SSL connections *may be* broken in platform code or Arduino core that 3.1.x uses.

> I then get an errorcode -9984 (invalid certificate or signature), thus the problem is somewhere in mbedtls_ssl_handshake() 

I am attempting to get more details on this and a reproduction, but since 3 users independently reported it I figured I should track it here.

chb60, 2021-04-04:
Hello, I have the same problem with my app, it works well with platformio 3.0.0 but make ssl error with upper version 3.1.x and 3.2.0 with this error [E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -1
All other code working well except this.
Thanks for your help

maxgerhardt, 2021-04-04:
In the linked topic it was found out that this is purely dependent on the Arduino-ESP32 core version. Core 1.0.4 is good but after that it' bad (1.0.5 and 1.0.6), and core 1.0.4 is used in Espressif32 platform 3.0.0, 1.0.5 in 3.1.x and 1.0.6 in 3.2.x. 

Or rather, as can be seen in https://github.com/espressif/arduino-esp32/issues/4992, "bad" in the sense that you now have to call an extra function `.setInsecure()` to get the old behavior again.

Not an issue in PlatformIO, hence closing.

closed 2021-04-04 by maxgerhardt