platformio/platform-espressif32 #517: ESP32 ULP-Toolchain: Build file not updated

nullptrdesaster, 2021-03-24:
Hi,
I imported the ulp_adc exaple code. It compiles flawless an runs on the ESP32.

However making a change in the ulp code in ‘ulp/adc.S’ is not always recognized by the toolchain and no new target is build.
Looking into file ‘/.pio/build/debug/esp-idf/main/ulp_main/adc.ulp.S’ reveals, that it is not updated.
Deleting it to force an update is not noticed by the toolchain. A rebuild succeeds after a clean run.

$pio platform update espressif32 
Platform espressif32
--------
Updating git+https://github.com/platformio/platform-espressif32 3.1.1+sha.0d3a4e2                  [Up-to-date]
Updating platformio/toolchain-xtensa32        2.50200.97 @ ~2.50200.0            [Up-to-date]
Updating platformio/toolchain-esp32ulp        1.22851.191205 @ ~1.22851.0        [Up-to-date]
Updating platformio/framework-arduinoespressif32 3.10005.210308 @ ~3.10005.0        [Up-to-date]
Updating platformio/framework-espidf          3.40200.210118 @ ~3.40200.0        [Up-to-date]
Updating platformio/tool-esptoolpy            1.30000.201119 @ ~1.30000.0        [Up-to-date]
Updating platformio/tool-openocd-esp32        2.1000.20201202 @ ~2.1000.0        [Up-to-date]
Updating platformio/tool-mkspiffs             2.230.0 @ ~2.230.0                 [Up-to-date]
Updating platformio/tool-cmake                3.16.4 @ ~3.16.0                   [Up-to-date]
Updating platformio/tool-ninja                1.7.1 @ ^1.7.0                     [Up-to-date]
 
Thanks for looking into this.

env:
debian 10.8, VSCode 1.54.3, Core 5.2.0a3, Home 3.3.4

nullptrdesaster, 2021-03-27:
What I did meanwhile:
The problem, that a changed  ulp\xx.S  code is not recompiled,  might be located in a malformed  "build.ninja" .
The assumption is based on these findings: 
1) The misbehavior is present in imported example-projects and in my own projects.
2) Projects are correctly recompiled  doing a  "idf.py build" in the esp-idf environment.
3) The recompilation is  controlled by "build.ninja". This step is differently handled from platformio as compared to esp-idf, as I found #includes within ulp\xx.S  code linking to other  ulp\yy.S files are handled differently. 

As I don't understand ninja at all, I'm stuck and can not contribute further.

Nagymadar, 2021-06-28:
Any progress or workaround yet?

nullptrdesaster, 2021-06-28:
Thanks for solving that. Works for me.

Nagymadar, 2021-06-28:
Deleting the whole directory each time is a workaround, _not a solution_. Any other idea?

nullptrdesaster, 2021-06-28:
Well, I made an update recently: 
pio upgrade --dev
pio update
pio platform update espressif32

and now for me the problem is solved. No need to to run a clean rebuild here anymore.

nullptrdesaster, 2021-06-28:
Oh -no -  I made a mistake. Switching between esp-idf and pio. Sorry. It still doesn't work for pio.

Nagymadar, 2021-06-28:
It isn't working for me either. So, anyone any ideas?

zkarcher, 2022-02-13:
Glad I'm not the only person experiencing this. I lost several nights to this issue, I thought my code was incorrect, but it turns out that changes to the .S file were being ignored.

I'm using a Makefile, so my workaround will be updating the make commands to delete compiled ULP objects before each build. Hopefully this will force a reload & rebuild each time.

closed 2022-08-10 by stale[bot]