platformio/platform-espressif32 #520: RTCWDT_RTC_RESET loop with 3.1.1

bradanlane, 2021-03-27:
I pulled down an old project of mine and compiled with the Platformio 5.1.1 with espressif32 3.1.1 and after flashing it wen into a reset loop with the RTCWDT_RTC_RESET message.

I removed and reinstalled the espressif32 3.1.1 "platform" through the platformio GUI within VS CODE. A clean compile, load, and same error.

Tried a simple "hello world" program and had the same problem.

Dropped back to epsressif32 2.1.0 and it compiled, loaded, and ran correctly.

I did not try to upgrade to see with release(s) from 2.1.0 to 3.1.1 may still work.

valeros, 2021-03-29:
Hi @bradanlane ! Does it work with the latest upstream version of the platform?

```ini
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
framework = arduino
board = esp32dev
```

bradanlane, 2021-03-29:
I'll test this in a few hours and report back.

bradanlane, 2021-03-29:
The Board is the TTGO T-Display (ESP32 520KB RAM, 4MB FLASH) and ST7789 240x135 display

PlatformIO 5.1.1

espressif32@2.1.0 works
espressif32@3.0.0 works
espressif32@3.1.0 does not work
espressif32@3.1.1 does not work
espressif32@3.2.0 does not work
upstream version does not work

**Error:**
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
```

valeros, 2021-03-29:
Does it work in Arduino IDE?

bradanlane, 2021-03-29:
I do not have an Arduino environment. I've been on PlatformIO + VS Code for the past 2 years.

It will be  while before I can set one up (_mostly because I have a deadline and 3.0.0 is working_).

rolandschmidt, 2021-04-30:
I have just run into the same problem.

I figured out, that there the issue is within using HSPI or at least if I comment the following line (and all other parts that are not useable without the HSPI) 

`SPIClass SPI1(HSPI);`

the code is compiling when using espressif32@3.2.0. I use the ESP32-Wrover-B with the 8Mb PSRAM.

When going back to `platform = espressif32@3.0.0` the complete code compiles without any issues.

drphil3d, 2021-08-03:
I'm having the same issue. 3.1.1 = fails to boot

drphil3d, 2021-08-03:
`rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:1
load:0x3fff0018,len:4
load:0x229c2132,len:-1052598751
1162 mmu set 00010000, pos 00010000
1162 mmu set 00020000, pos 00020000
1162 mmu set 00030000, pos 00030000
1162 mmu set 00040000, pos 00040000
1162 mmu set 00050000, pos 00050000
1162 mmu set 00060000, pos 00060000
1162 mmu set 00070000, pos 00070000
1162 mmu set 00080000, pos 00080000
1162 mmu set 00090000, pos 00090000
1162 mmu set 000a0000, pos 000a0000
1162 mmu set 000b0000, pos 000b0000
1162 mmu set 000c0000, pos 000c0000
1162 mmu set 000d0000, pos 000d0000
1162 mmu set 000e0000, pos 000e0000
1162 mmu set 000f0000, pos 000f0000
1162 mmu set 00100000, pos 00100000
1162 mmu set 00110000, pos 00110000
ets Jun  8 2016 00:22:57`

Flamy69, 2021-09-03:
> The Board is the TTGO T-Display (ESP32 520KB RAM, 4MB FLASH) and ST7789 240x135 display
>
> PlatformIO 5.1.1
>
> espressif32@2.1.0 works
> espressif32@3.0.0 works
> espressif32@3.1.0 does not work [...]

Same issue and same test results. Board NodeMCU ESP-32S and board = esp32dev.

BastelBaus, 2022-01-23:
For me as well, even if i just declare a pointer to it, strange ... 
SPIClass* SPI1 = NULL;

Same hier: https://community.platformio.org/t/esp32-does-not-boot-on-espressif32-3-3-2/23646/6

Anyone found a solution ?

BastelBaus, 2022-01-25:
I continued debugging ... empty class with constructure and nothing else ... still same behaviour. So my interpretation it is not the SPI file itself which is buggy ... I edited also the main.cpp and found that the effect must take place before the start of the main function (the real one :-)). So probably some code within class initializer or something ... I feal really stupid that I did not get this to running ... any Ideas ??

BastelBaus, 2022-01-25:
I think I got it! 
It is related to the variable name SPI1. Any other name works for me. I grepped the ESP sources (1.0.6) and found only one possible position:
  Arduino15\packages\esp32\hardware\esp32\1.0.6\tools\sdk\include\soc\soc\spi_struct.h
	extern spi_dev_t SPI1;
  Arduino15\packages\esp32\hardware\esp32\1.0.6\tools\sdk\ld\esp32.peripherals.ld
	Zeile 2: PROVIDE ( SPI1 = 0x3ff42000 );

So w/o searching the manuals I assume that is the build in code (ROM?) of the ESP32. If this is correct, The only thing which I did not get, why I did not get a compile or at least link time error .... anyone has an idea ?
BR, 
Bastel Baus

valeros, 2022-05-04:
Should be fixed in the latest version of the dev-platform. Please reopen if the problem persists.

closed 2022-05-04 by valeros