platformio/platform-espressif32 #522: add arduino-idf-v4.3

atecroby, 2021-03-29:
Has anyone succeeded to upgrade arduino-idf to v4.3 or current version? (or is there any plan to do it next weeks?)

There's a buffer underrun on A2DP_sink that has been corrected on idf v4.3 so I would need updated libraries.

I'm trying to use esp32-arduino-lib-builder without success

Thank you

valeros, 2021-03-29:
Hi @atecroby ! We support only the stable versions of frameworks. ESP-IDF `v4.3` hasn't been released yet. You can try to override the default package using the [platform_packages](https://docs.platformio.org/en/latest/projectconf/section_env_platform.html#platform-packages) option, but there is no guarantee that it'll work.

closed 2021-03-29 by valeros

helaagardabbou, 2021-06-17:
ESP-IDF v4.3 has been released 2 days ago. when platformio will support it  please ?