platformio/platform-espressif32 #524: ardiuno

maxgerhardt, 2021-04-04:
Per https://github.com/espressif/arduino-esp32/issues/5007 and https://github.com/espressif/arduino-esp32/issues/4998

The `espidf-arduino-blink` is uncompilable because the referenced `#idf-release/v4.0` was killed in the repo.

https://github.com/platformio/platform-espressif32/blob/b9530d31e4f8dfe00b31994c70b1ac96d82346cb/examples/espidf-arduino-blink/platformio.ini#L16-L18

[image]

sguarin, 2021-04-12:
Same problem here. If I use master branch I also got:

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.2.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 0.0.0+sha.371f382 
 - framework-espidf 3.40001.200521 (4.0.1) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - tool-ninja 1.7.1 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32 2.80400.210211 (8.4.0)
Reading CMake configuration...
-- Project version: e11a3b7
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also "/home/sguarin/MEGA/Electronics/ESP32/src/espidf-arduino-blink/.pio/build/stable/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /home/sguarin/.platformio/packages/framework-espidf/tools/cmake/build.cmake:185 (message):
  Failed to resolve component 'arduino_tinyusb'.
Call Stack (most recent call first):
  /home/sguarin/.platformio/packages/framework-espidf/tools/cmake/build.cmake:216 (__build_resolve_and_add_req)
  /home/sguarin/.platformio/packages/framework-espidf/tools/cmake/build.cmake:425 (__build_expand_requirements)
  /home/sguarin/.platformio/packages/framework-espidf/tools/cmake/project.cmake:348 (idf_build_process)
  CMakeLists.txt:3 (project)
```

maxgerhardt, 2021-04-12:
Does the workaround [here](https://github.com/espressif/arduino-esp32/issues/4998#issuecomment-813022103) work?

sguarin, 2021-04-12:
Well I will gave this a try. Besides I will be stucked in that commit (https://github.com/marcovannoord/arduino-esp32.git), it gives me the error that requires another component:
```
  Failed to resolve component 'ethernet'.
Call Stack (most recent call first):                                                                                                    
  /home/sguarin/.platformio/packages/framework-espidf/tools/cmake/build.cmake:211 (__build_resolve_and_add_req)                         
  /home/sguarin/.platformio/packages/framework-espidf/tools/cmake/build.cmake:425 (__build_expand_requirements)                         
  /home/sguarin/.platformio/packages/framework-espidf/tools/cmake/project.cmake:348 (idf_build_process)                                 
  CMakeLists.txt:3 (project) 
```

The problem is that arduino-esp32 now has support for esp32s2 and incorporates another component arduino_tinyusb for usb support.
I chat a little in arduino-esp32 channel, and lberstone point me in the espressif/esp32-arduino-lib-builder direction. There are some scripts for building arduino-esp32 to work as idf component.
It seems platformio-espressif32 needs some work here.

EmilianoQS, 2021-04-19:
Is there any solution to this problem? 

Can't make it work, tried the workaround mentioned by @maxgerhardt but I get problems with other libraries.

Jason2866, 2021-04-21:
This works for IDF4.4
```
platform                    = https://github.com/Jason2866/platform-espressif32.git#feature/arduino-idf-v4.4
platform_packages           = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
```

yogyyoghurt, 2021-04-25:
Hi everybody! 
I also have problems building the espidf-arduino-blink example.
Basicly the referenced #idf-release/v4.0 could not be found in the repo. Then I played around with a number of modifications. Latest was Jason2866' approach. But still need a number of modifications for problems with 'arduino_tinyusb', 'main' and 'esp_littlefs'. I tried to eliminate them in CMakeLists.txt. But this lead to more problems and is for sure not the right way. 
Since I don't have detailed knowledge about the whole build process, my question is: Is there an actual workaround to get that example running? Or any other examples using espidf together with arduino framework?

Jason2866, 2021-04-25:
@yogyyoghurt  Just tried, compiled and working with ESP32 and ESP32-S2
You can find my test setup https://github.com/Jason2866/arduino-blink

yogyyoghurt, 2021-04-25:
> 
After doing my very first projects in arduino framework, I would like to test some more professional issues, which seem only be possible under espidf framework. So, as I noticed from a number of discussions (e.g. https://community.platformio.org/t/using-esp-idf-library-within-the-arduino-framework-esp32/17226) this should be possible in a very handy way using both frameworks.

@Jason2866 When I have a look in your platformio.ini, it just uses framework Arduino. Right? So this discussion here is about the espidf-arduino-blink example not arduino-blink. Or didn't I get some details?

> 
> Does the workaround [here](https://github.com/espressif/arduino-esp32/issues/4998#issuecomment-813022103) work?

@maxgerhardt I have tried this. Seems to work!!! Compiling and linking is OK. I didn't test the application so far. But by the way I have a more general question. Using such a more 'private' branch of a platform_package doesn't make me really happy, although it works. Is it possible to tell platformio, not to check online if this branch still exists? So I would like to stay and ensure not to destroy my working environment?

sguarin, 2021-04-26:
> @yogyyoghurt Just tried, compiled and working with ESP32 and ESP32-S2
> You can find my test setup https://github.com/Jason2866/arduino-blink

Thanks. Your test setup works fine, but the problem we are discussing here is when you use:
```
framework = arduino, espidf
```

Jason2866, 2021-04-26:
Should work too, it uses the same setup as https://github.com/espressif/arduino-esp32/issues/4998#issuecomment-813022103
Just add espidf to the framework

Jason2866, 2021-04-26:
> @maxgerhardt I have tried this. Seems to work!!! Compiling and linking is OK. I didn't test the application so far. But by the way I have a more general question. Using such a more 'private' branch [...]

Just fork the setup (platform_packages) which is working for you and refer to your own.
As long there is no stable version released, you could not expect a setup will last for early alpha or beta stuff.

apwwn, 2021-04-26:
maybe the definition of board need to be updated as the latest expressif package is officially supporting esp32s2 : [https://github.com/espressif/arduino-esp32#esp32-s2-and-esp32-c3-support](url)

(all branches used for esp32s2 has been deleted by expressif)

Jason2866, 2021-04-26:
@apwwn  the branch esp32s2 is now integrated in master and so not needed anymore. You can easily define boards yourself. Just place your `boards.json` in folder `boards` in root of Platformio project. I have done this for project [Tasmota](https://github.com/arendst/Tasmota)

maxgerhardt, 2021-05-23:
@valeros people are still hitting this issue, see e.g. [community topic](https://community.platformio.org/t/espressif-32-platform-the-staging-libraries/21585). Please update the branches / build system of the Arduino+ESP-IDF examples? It's indeed bad that Arduino-ESP32 deletes branches at any time, so maybe a backup fork would be good 🤔