platformio/platform-espressif32 #526: Error tool-mkspiffs

joelsernamoreno, 2021-04-05:
Hi!

When I try to use platformio it cannot install tool-mkspiffs. A few months ago this was working, but now it has stopped working.

The error is as follows:

Processing ESP32 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing tool-mkspiffs @ ~2.230.0
**Error: Detected unknown package 'tool-mkspiffs'**

Thanks

valeros, 2021-04-05:
Hi @joelsernamoreno ! Could you please run two commands in your terminal:
 `pio system info`
`pio platform show espressif32`

joelsernamoreno, 2021-04-05:
Hi!

pio system info:
/home/joel/Música/pio-system.png

pio platform show espressif32: 
[platform.txt](https://github.com/platformio/platform-espressif32/files/6257179/platform.txt)

I fixed this error, read my final comment at https://github.com/rfquack/RFQuack/issues/6

Thanks!

closed 2021-04-05 by valeros

TIMESTICKING, 2022-10-04:
Hi, I got the same issue here. It worked which means can upload in a few weeks ago. but today when I tried to upload the program it starts to
```
Tool Manager: Installing platformio/tool-mkspiffs @ ~2.230.0
Error: 
```
It doesnt give me any message abt what the error is, and also Im curious 
1. why would this suddenly requires the tool-mkspiffs, why it didnt pop up in a few weeks ago.
2. when I do `pio pkg show espressif32`, it also gave an `Error`, is my `espressif32` package broken?
3. IF it is, how do I upgrade espressif32, bc  `pio pkg update` also gives an `Error`... :(