platformio/platform-espressif32 #527: Core panic error and bootloop with mcpwm_capture_signal_get_edge inside an interceptor handler

ifynk, 2021-04-05:
**Configuration**

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.2.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40200.210118 (4.2.0) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - tool-ninja 1.7.1 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32 2.80400.210211 (8.4.0)
```

**Code example**

```
#include "driver/mcpwm.h"
#include "soc/mcpwm_periph.h"

#define GPIO_CAP0_IN 23
#define CAP0_INT_EN BIT(27)

static mcpwm_dev_t *MCPWM[2] = {&MCPWM0, &MCPWM1};

static void IRAM_ATTR isrHandler() {
  uint32_t mcpwm_intr_status = MCPWM[MCPWM_UNIT_0]->int_st.val;

  if (mcpwm_intr_status & CAP0_INT_EN) {
    uint32_t edge = mcpwm_capture_signal_get_edge(MCPWM_UNIT_0, MCPWM_SELECT_CAP0);
  }

  MCPWM[MCPWM_UNIT_0]->int_clr.val = mcpwm_intr_status;
}

void app_main() {
  mcpwm_gpio_init(MCPWM_UNIT_0, MCPWM_CAP_0, GPIO_CAP0_IN);
  gpio_pulldown_en(GPIO_CAP0_IN);

  mcpwm_capture_enable(MCPWM_UNIT_0, MCPWM_SELECT_CAP0, (MCPWM_POS_EDGE | MCPWM_NEG_EDGE), 0);

  MCPWM[MCPWM_UNIT_0]->int_ena.val = CAP0_INT_EN;
  mcpwm_isr_register(MCPWM_UNIT_0, isrHandler, NULL, ESP_INTR_FLAG_IRAM, NULL);
}
```

**Stack trace**
```
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400e2a1b  PS      : 0x00060031  A0      : 0x800d0e82  A1      : 0x3ffb0780  
A2      : 0x3ffb0034  A3      : 0x0a33062c  A4      : 0x00000000  A5      : 0x3ffb07b0  
A6      : 0x0000cdcd  A7      : 0x00000000  A8      : 0x08000000  A9      : 0x3ff5e000  
A10     : 0x00000001  A11     : 0x00060023  A12     : 0x3ffb4840  A13     : 0x3ffb4810  
A14     : 0x00000000  A15     : 0x40080e94  SAR     : 0x00000005  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x400e2a18:0x3ffb0780 0x400d0e7f:0x3ffb07a0 0x40080ea9:0x3ffb07e0 0x40082011:0x3ffb0800 0x4000bfed:0x3ffb4770 0x40084815:0x3ffb4780 0x400d1983:0x3ffb47a0 0x400d19e1:0x3ffb47e0 0x400d0ef9:0x3ffb4810 0x400d0a33:0x3ffb4840 0x400d123a:0x3ffb4860 0x400845bd:0x3ffb4890
  #0  0x400e2a18:0x3ffb0780 in mcpwm_ll_get_capture_val at /home/ifynk/.platformio/packages/framework-espidf/components/soc/src/esp32/include/hal/mcpwm_ll.h:671
      (inlined by) mcpwm_hal_capture_get_result at /home/ifynk/.platformio/packages/framework-espidf/components/soc/src/hal/mcpwm_hal.c:217
  #1  0x400d0e7f:0x3ffb07a0 in mcpwm_capture_signal_get_edge at /home/ifynk/.platformio/packages/framework-espidf/components/driver/mcpwm.c:585 (discriminator 2)
  #2  0x40080ea9:0x3ffb07e0 in isrHandler at src/main.c:13
  #3  0x40082011:0x3ffb0800 in _xt_lowint1 at /home/ifynk/.platformio/packages/framework-espidf/components/freertos/xtensa/xtensa_vectors.S:1105
  #4  0x4000bfed:0x3ffb4770 in ?? ??:0
  #5  0x40084815:0x3ffb4780 in vPortExitCritical at /home/ifynk/.platformio/packages/framework-espidf/components/freertos/xtensa/port.c:419
  #6  0x400d1983:0x3ffb47a0 in esp_intr_alloc_intrstatus at /home/ifynk/.platformio/packages/framework-espidf/components/esp32/intr_alloc.c:673
  #7  0x400d19e1:0x3ffb47e0 in esp_intr_alloc at /home/ifynk/.platformio/packages/framework-espidf/components/esp32/intr_alloc.c:693
  #8  0x400d0ef9:0x3ffb4810 in mcpwm_isr_register at /home/ifynk/.platformio/packages/framework-espidf/components/driver/mcpwm.c:620 (discriminator 2)
  #9  0x400d0a33:0x3ffb4840 in app_main at src/main.c:26
  #10 0x400d123a:0x3ffb4860 in main_task at /home/ifynk/.platformio/packages/framework-espidf/components/esp32/cpu_start.c:600 (discriminator 2)
  #11 0x400845bd:0x3ffb4890 in vPortTaskWrapper at /home/ifynk/.platformio/packages/framework-espidf/components/freertos/xtensa/port.c:143

ELF file SHA256: 02acffaa9e841c45

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7268
load:0x40078000,len:13212
load:0x40080400,len:4568
entry 0x400806f4
I (29) boot: ESP-IDF 3.40200.210118 2nd stage bootloader
I (29) boot: compile time 11:19:07
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x067ec ( 26604) map
I (122) esp_image: segment 1: paddr=0x00016814 vaddr=0x3ffb0000 size=0x021c4 (  8644) load
I (126) esp_image: segment 2: paddr=0x000189e0 vaddr=0x40080000 size=0x00404 (  1028) load
I (131) esp_image: segment 3: paddr=0x00018dec vaddr=0x40080404 size=0x0722c ( 29228) load
I (152) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x132ac ( 78508) map
I (182) esp_image: segment 5: paddr=0x000332d4 vaddr=0x40087630 size=0x028a8 ( 10408) load
I (193) boot: Loaded app from partition at offset 0x10000
I (193) boot: Disabling RNG early entropy source...
I (194) cpu_start: Pro cpu up.
I (197) cpu_start: Application information:
I (202) cpu_start: Project name:     iebus
I (207) cpu_start: App version:      1
I (211) cpu_start: Compile time:     Apr  5 2021 11:18:42
I (218) cpu_start: ELF file SHA256:  02acffaa9e841c45...
I (224) cpu_start: ESP-IDF:          3.40200.210118
I (229) cpu_start: Starting app cpu, entry point is 0x400810bc
I (221) cpu_start: App cpu up.
I (240) heap_init: Initializing. RAM available for dynamic allocation:
I (247) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (253) heap_init: At 3FFB29F0 len 0002D610 (181 KiB): DRAM
I (259) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (265) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (272) heap_init: At 40089ED8 len 00016128 (88 KiB): IRAM
I (278) cpu_start: Pro cpu start user code
I (296) spi_flash: detected chip: generic
I (297) spi_flash: flash io: dio
I (297) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
```

valeros, 2021-04-05:
Hi @ifynk. Unfortunately it's not that easy to figure out whether it's the problem with the ESP-IDF integration within PlatformIO or with ESP-IDF itself. It would be great if you could just compile and upload the same project using the native ESP-IDF tools.

valeros, 2022-04-18:
Closing as no further info provided. Feel free to reopen the issue if you're still experiencing the problem with the latest upstream version of the platform.

closed 2022-04-18 by valeros