platformio/platform-espressif32 #531: Requesting Support For ESP32-S0WD Arduino

mlpzaq9998, 2021-04-16:
ESP32-S0WD is Single core processor

Chip is unknown ESP32 (revision 1)
Features: WiFi, BT, Single Core, 240MHz, VRef calibration in efuse, Coding Scheme 3/4
Crystal is 40MHz

https://github.com/lbernstone/arduino-esp32-solo

Jason2866, 2021-04-20:
I did a custom build. Package published today, should be in registry of platformio soon.
Will be available with:
```
platform_packages           = tasmota/framework-arduinoespressif32@3.10006.210420
```

You can use now:
```
platform_packages           = framework-arduinoespressif32 @ https://github.com/Jason2866/esp32-arduino-lib-builder/raw/framework-arduinoespressif32/framework-arduinoespressif32-solo1-release_v3.3-cc43307dc.tar.gz
```

mlpzaq9998, 2021-04-21:
Thank for your work

mlpzaq9998, 2021-04-21:
Thank for your work