platformio/platform-espressif32 #532: Is it possible to use an #ifdef to distinquish between versions

Bascy, 2021-04-16:
We have several projects using espressif32 platform and we don't update to the latest version automatically.
We are now running into a problem with a shared piece of code between project which are build on version 1.12.4 and 3.2.0 where the signature of a method has changes. 

I don;t want to update the 1.12.4 project just yet so I need to have a way to determine the current version of espressif32 platform that is being used.

Is there a define I can use in an #ifdef to realize that?

valeros, 2021-04-19:
Hi @Bascy ! I suppose you'd better check the version of the underlying framework instead of the platform. Anyway, there is no such a define, but it can be easily added using [an extra script](https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html):
```python
Import ("env")

from platformio.package import version

platform = env.PioPlatform()

env.Append(
    CPPDEFINES=[
        ("ESPRESSIF32_PLATFORM_VERSION", platform.version.replace(".", "0")),
        (
            "ESPRESSIF32_ARDUINO_VERSION",
            version.get_original_version(
                platform.get_package_version("framework-arduinoespressif32")
            ).replace(".", "0"),
        ),
    ]
)
```

closed 2021-04-19 by valeros

Bascy, 2021-04-19:
Thanks for the suggestion