platformio/platform-espressif32 #533: IllegalInstruction panic when using the ADC

Kampi, 2021-04-17:
I use the esp-IDF together with Arduino in my LoRa-Project and my application crash with the following message:

```
�D (837) intr_alloc: Connected src 34 to int 3 (cpu 1)
[E][esp32-hal-cpu.c:125] removeApbChangeCallback(): not found func=400D7DD0 arg=3FFBDCB8
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400d5a3c: 0020c0ea ffffffff ffffffff
Core 1 register dump:
PC      : 0x400d5a40  PS      : 0x00060130  A0      : 0x800d5b1b  A1      : 0x3ffb1ef0
A2      : 0x00000021  A3      : 0x3ffc0194  A4      : 0x00000010  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x00000009  A9      : 0xffffffff
A10     : 0x3ffc0194  A11     : 0x00000004  A12     : 0x3ffc01a3  A13     : 0x0000008c
A14     : 0x00000004  A15     : 0x00060023  SAR     : 0x00000011  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

ELF file SHA256: 8837833167fb16d8

Backtrace: 0x400d5a3d:0x3ffb1ef0 0x400d5b18:0x3ffb1f10 0x400d1198:0x3ffb1f40 0x400d1639:0x3ffb1f70 0x400d0f53:0x3ffb1fa0 0x40086079:0x3ffb1fc0
  #0  0x400d5a3d:0x3ffb1ef0 in __adcAttachPin at C:\Users\Danie\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-adc.c:102
  #1  0x400d5b18:0x3ffb1f10 in __analogRead at C:\Users\Danie\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-adc.c:137
  #2  0x400d1198:0x3ffb1f40 in Transmit(osjob_t*) at src/application/application.cpp:89
  #3  0x400d1639:0x3ffb1f70 in setup() at src/application/application.cpp:301
  #4  0x400d0f53:0x3ffb1fa0 in loopTask(void*) at src/main.cpp:24
  #5  0x40086079:0x3ffb1fc0 in vPortTaskWrapper at C:\Users\Danie\.platformio\packages\framework-espidf\components\freertos/port.c:143

Rebooting...
ets Jun  8 2016 00:22:57
```

I use the following `main` to start the application

```
#include <Arduino.h>
#include <soc/soc.h>
#include <soc/rtc_cntl_reg.h>
#include <soc/timer_group_reg.h>
#include <soc/timer_group_struct.h>

#include <freertos/task.h>
#include <freertos/FreeRTOS.h>
#include <freertos/event_groups.h>

#include "blufi/blufi.h"
#include "utils/ulp_utils.h"
#include "misc/DebugPrint.h"
#include "misc/ResetHandle.h"

#include <ulp_main.h>

#ifndef CONFIG_ARDUINO_LOOP_STACK_SIZE
    #define CONFIG_ARDUINO_LOOP_STACK_SIZE          8192
#endif

static void loopTask(void* p_Parameters)
{
    setup();

    while(1)
    {
        TIMERG0.wdt_wprotect = TIMG_WDT_WKEY_VALUE;
        TIMERG0.wdt_feed = 1;
        TIMERG0.wdt_wprotect = 0;

        micros();
        loop();
    }
}

#ifdef APP_WITH_BLUFI
    static void BluFi_DisconnectHandler(void)
    {
        // Restart the ESP to reset the system into a known state and to apply the changes
        esp_restart();
    }
#endif

static void StartApplication(void)
{
    DEBUG_INFO("[INFO] Starting application...");
    initArduino();
    xTaskCreatePinnedToCore(loopTask, "loopTask", CONFIG_ARDUINO_LOOP_STACK_SIZE, NULL, 1, NULL, CONFIG_ARDUINO_RUNNING_CORE);
}

extern "C" void app_main(void)
{
    // Disable the brownout detector
    WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0);

    if(ResetHandle_isPowerOnReset())
    {
        DEBUG_INFO("[INFO] Power-On-Reset occured!");

        #ifdef APP_WITH_BLUFI
            DEBUG_INFO("[INFO] Starting BluFi mode...");
            BluFi_Init(BluFi_DisconnectHandler);
        #else
            StartApplication();
        #endif
    }
    else if(ResetHandle_isDeepSleepReset())
    {
        DEBUG_INFO("[INFO] Deep sleep Reset occured!");
        StartApplication();
    }
    else
    {
        DEBUG_INFO("[INFO] Unknown reset!");
        ResetHandle_Print(rtc_get_reset_reason(0));
        ResetHandle_Print(rtc_get_reset_reason(1));

        DEBUG_INFO("[INFO] Switch back to sleep mode...");
        esp_sleep_enable_timer_wakeup(60ULL * 1000000ULL);
        esp_deep_sleep_start();
    }
}
```

The application crashes at the following point:

```
static void Transmit(osjob_t* p_Job)
{
    if(LMIC.opmode & OP_TXRXPEND)
    {
        DEBUG_INFO("[INFO] OP_TXRXPEND, not sending");
    }
    else
    {
        float Battery;
        float Moisture;

        digitalWrite(BAT_E, LOW);
        digitalWrite(SENSOR_E, HIGH);

        // Get the battery voltage
        Crash ---> Battery = analogRead(ANALOG_BATTERY) * 3.3 / 4095.0 * 2;

        // Get the moisture sensor voltage
        Moisture = analogRead(ANALOG_MOISTURE) * 3.3 / 4095.0;

        digitalWrite(SENSOR_E, LOW);
        digitalWrite(BAT_E, HIGH);

        sprintf(Buffer, "{\"Battery\": %f, \"Moisture\": %f}", Battery, Moisture);

        DEBUG_INFO("[INFO] Battery: %f", Battery);
        DEBUG_INFO("[INFO] Humidity: %f", Moisture);

        LMIC_setTxData2(1, (unsigned char*)Buffer, sizeof(Buffer), 0);
    }
}
```

What´s the reason for this crash?

valeros, 2021-04-19:
Hi @Kampi ! Unfortunately, it's not that easy to resolve such problems. Have you tried to attach a debug probe and look what's happening inside that function? Also, this link might be useful https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#illegal-instr-msg

Kampi, 2021-04-19:
Hi @valeros,

I will check if I can use a debug probe in that design and then I will try it.

closed 2021-05-06 by Kampi