platformio/platform-espressif32 #534: SD Card write stability worse since Version 3.0.0

savejeff, 2021-04-18:
Hi, 

I'm working code that writes Loggings to an SD Card using the ESP32. 
The code works very stable for more than a year now. but since the introduction of Version 3.1.0 and higher (including 3.2.0) the 
SD Card interaction is not reliable anymore. 

It took me quite a lot of time to get sure about this. The Problem only appears after multiple hours of writing. 
With 3.0.0 I have never problems writing to a file for more than 44h. With newer versions of the file.write function returns 0 and writing to the sd card requires reboot. 

My first question is: where should I post this problem? I'm thinking about the espressif forum, but I use the Arduino Firmware in PlatformIO. Is the Espressif Firmware versioning identical to the Arduino Core Version?

I see this unreliability as a serious matter. not only that I can not rely on the SD Card logging but in general its quite a core feature the ESP32 architecture. As this is quite hard to track down and happens so rarely I'm afraid it will not get picked up by the dev's as a serious matter to fix. any tips?

I'm working on an example code that uses simple operations like open and write from a static buffer that crashes after long time with 3.2.0 but not with 3.0.0

valeros, 2021-04-19:
Hi @savejeff ! Have you tried to compile your project using the latest Arduino Core for ESP32 v1.0.6 in the Arduino IDE? Does it work there reliably? In a nutshell, we use the same stable packages, but the build process may slightly differ.

savejeff, 2021-04-20:
Hi!

I'm currently trying to reduce the code to only the essential parts. 
basic "just write over and over in an a loop" does not produce the error. 
In my project I'm using two tasks pinned to each core of the ESP32. A Threadsafe Ringbuffer is used to transfer data that is to be written to the Logfile on the SD Card. 

I'm pretty sure its a problem with the Espressif Framework and not with the platformIO wrapper framework. 

I'll post an example code I can reliably reproduce the error on 3.2.0 and runs correctly on 3.0.0. It just takes very long to test it because crashes only happen after some hours if at all

valeros, 2022-04-18:
Closing as no further information was provided. Feel free to reopen the issue if you are still experiencing the problem with the upstream version of the platform.

closed 2022-04-18 by valeros

KAMRANRASOOL777, 2023-01-22:
Hello everyone.
I am also having this issue and maybe someone had got some Idea.
I am using Arduino framework version 2.0.6. I get this message as warning "[sd_diskio.cpp:186] sdCommand(): token error [13] 0x3". After this, logging of data is stopped despite write command being sent.
Any help will be appreciated. Thanks

savejeff, 2023-01-22:
I tried a lot of things to fix the problem. 
In my experience on the ESP32 the SD card write is very reliable. I only have problems with the ESP32-S3 where I have to use a lower clock rate of 10Mhz to avoid infrequent weite errors.

My experience is mainly based on the SDfat library and not the inbuilt SD.h lib. I would highly recommend using the FSfat, as it is faster and more feature rich

KAMRANRASOOL777, 2023-01-22:
Thanks @savejeff 
I am using a custom circuit board for this. But I will try your suggestion for FSfat library and will post my results.

KAMRANRASOOL777, 2023-02-05:
The solution to this problem which I found is to keep on checking the return number of write function. If the return is 0, just close the file and again open it with the APPEND attribute. It might not be a sober solution but OK to me for the time being.