platformio/platform-espressif32 #537: fatal error: esp_netif.h: No such file or directory

johnsmakar, 2021-05-02:
Hello,

I am trying to run WiFi on my esp and I am using the ASyncUDP library from espressif but I get 
``` fatal error: esp_netif.h: No such file or directory ```

A platformIO forum recommended I do:
```
platform_packages = 
    framework-espidf @ https://github.com/espressif/esp-idf.git#release/v4.1
```

but this is what I get:

```
Platform Manager: Installing git+https://github.com/platformio/platform-espressif32.git
Error: Traceback (most recent call last):
  File "c:\users\john\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 70, in process
    p = PlatformFactory.new(self.options["platform"])
  File "c:\users\john\.platformio\penv\lib\site-packages\platformio\platform\factory.py", line 54, in new
    raise UnknownPlatform(pkg_or_spec)
platformio.platform.exception.UnknownPlatform: Unknown development platform 'https://github.com/platformio/platform-espressif32.git'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\john\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "c:\users\john\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\john\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\john\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\john\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\john\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\john\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\john\.platformio\penv\lib\site-packages\click\decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\john\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 133, in cli
    process_env(
  File "c:\users\john\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 175, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "c:\users\john\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 72, in process
    self.cmd_ctx.invoke(
  File "c:\users\john\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\john\.platformio\penv\lib\site-packages\platformio\commands\platform.py", line 325, in platform_install
    pkg = pm.install(
  File "c:\users\john\.platformio\penv\lib\site-packages\platformio\package\manager\platform.py", line 51, in install
    pkg = super(PlatformPackageManager, self).install(
  File "c:\users\john\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 48, in install
    pkg = self._install(
  File "c:\users\john\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97, in _install
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "c:\users\john\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 141, in install_from_url
    vcs = VCSClientFactory.new(tmp_dir, url)
  File "c:\users\john\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 54, in new
    obj = getattr(sys.modules[__name__], "%sClient" % type_.title())(
  File "c:\users\john\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 137, in __init__
    self.configure()
  File "c:\users\john\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 146, in configure
    result = proc.exec_command([cls.command, "--exec-path"])
  File "c:\users\john\.platformio\penv\lib\site-packages\platformio\proc.py", line 117, in exec_command
    p = subprocess.Popen(*args, **kwargs)
  File "C:\Users\John\.platformio\python3\lib\subprocess.py", line 947, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\John\.platformio\python3\lib\subprocess.py", line 1416, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

I ran pip install to get the latest install, but no hope.

Any other recommendations would be really helpful!

valeros, 2021-05-03:
Do you have Git application installed in your system?

johnsmakar, 2021-05-03:
Yes, I do.

johnsmakar, 2021-05-09:
UPDATE: Resolved, the there is a `ESPASyncUDP` and `ASyncUDP` , we were using the ESP version which was built for the espressif8266 Platform while the standard ASyncUDP is for espressif32.

closed 2021-05-09 by johnsmakar

vietanhnb288, 2022-07-26:
@johnsmakar  hi bro can you show me the way you fix this problem? I'm still get trouble with it. Thank you very much
My error:    fatal error: esp_netif.h: No such file or directory

typetrade, 2024-10-18:
Header File[](https://docs.espressif.com/projects/esp-idf/zh_CN/v5.3.1/esp32/api-reference/network/esp_netif.html#header-file)
[components/esp_netif/include/esp_netif.h](https://github.com/espressif/esp-idf/blob/v5.3.1/components/esp_netif/include/esp_netif.h)

This header file can be included with:

`#include "esp_netif.h"`

This header file is a part of the API provided by the esp_netif component. To declare that your component depends on esp_netif, add the following to your CMakeLists.txt:

`REQUIRES esp_netif`

or

`PRIV_REQUIRES esp_netif`