platformio/platform-espressif32 #540: TTGO T-Camera support needed

mottibz, 2021-05-10:
Hello, I am using PlatformIO with VSCode for some time (developing on the ESP32) and very happy with it, thanks!
I need to develop code for the TTGO T-Camera (regular board, not the Plus) and can't find a way to define it for a new project.
Can you please help me add support for that board?
Thanks a lot!

valeros, 2021-05-10:
What framework do you plan to use with this board?

mottibz, 2021-05-10:
I am using the Arduino framework.
Thanks a lot for your help!

valeros, 2021-05-10:
There is no official variant for this board, so you need to add it to this repo https://github.com/espressif/arduino-esp32 in the first place.

mottibz, 2021-05-10:
Hi, I am not familiar with the process of doing that. Can you help me with that? I will very highly appreciate that!