platformio/platform-espressif32 #542: rework to use esp-idf v4.4+

enriquewph, 2021-05-12:
It currently is a pain in the a** to get arduino working as a component for esp-idf,
We should work on a rework to integrate the building tools from esp-idf v4.0+ more natively, and document this use-case in a better way.

ideally, can we get a development version using arduino-esp32 v2.0.0-dev, also using a somewhat recent esp-idf 4.4 version?

I'm happy to contribute to this, i just need to know, where to start..

dumarjo, 2021-06-02:
we are stuck with 4.0 only of esp-idf+arduino. I whish to be able to use the newest esp-idf also.

konstantingoretzki, 2021-08-03:
Any update on this?
The current way to use `esp-idf` with `arduino-esp32` as an component involves using [this "old" fork](https://github.com/platformio/platform-espressif32/commit/56f9c60081fadd2100330732ee27691981be5bb9) from Feb 21.

Btw +1 for a needed documentation.
Without stumbling into [espidf-arduino-blink](https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-arduino-blink) I could not find any docu about the possibility to do this using PlatformIO.

closed 2022-08-10 by stale[bot]