platformio/platform-espressif32 #543: Can't run 'pio run -t menuconfig'

silardg, 2021-05-18:
I want to setup OTA to http via the menuconfig.
I'm running this platformio.ini:
`
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
monitor_speed = 115200
monitor_filters = esp32_exception_decoder, colorize, time
monitor_flags = --raw
board_build.partitions = partitions_custom.csv
lib_deps = 
	knolleary/PubSubClient @ ^2.8
	256dpi/MQTT@^2.5.0
	adafruit/Adafruit NeoPixel@^1.8.0
	kitesurfer1404/WS2812FX@^1.3.3

`

On 'pio run -t menuconfig':
`
❯ pio run -t menuconfig
Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (3.2.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <PubSubClient> 2.8.0
|-- <MQTT> 2.5.0
|-- <Adafruit NeoPixel> 1.8.0
|-- <WS2812FX> 1.3.3
|   |-- <Adafruit NeoPixel> 1.8.0
|-- <Constants>
|-- <NVS>
|   |-- <Constants>
|   |-- <Preferences> 1.0
|-- <Network>
|   |-- <Constants>
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <MQTT> 2.5.0
|   |-- <NVS>
|   |   |-- <Constants>
|   |   |-- <Preferences> 1.0
|   |-- <WebServer> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <FS> 1.0
|   |-- <WiFi> 1.0
Building in release mode
scons: *** Do not know how to make File target `menuconfig' (/Users/silardg/Documents/Development/evergreen/evergreenFirmware/evergreen/menuconfig).  Stop.
`

valeros, 2021-05-19:
You can use the `menuconfig` target only with ESP-IDF framework.

closed 2021-05-19 by valeros

zackees, 2023-01-05:
Literally your tool says to run `pio run -t menuconfig`