platformio/platform-espressif32 #545: esp32_exception_decoder do not decode with option time.

GOB52, 2021-05-21:
Vscode 1.56.2
PlatformIO Core 5.1.1 Home 3.3.4
PLATFORM Espressif 32 (3.2.0)
BOARD M5Stack Grey
FRAMEWORK arduino

monitor_filters = esp32_exception_decoder
decoded.

monitor_filters = esp32_exception_decoder,time
not decoded.

time is processed first, and the timestamped text is passed to filter_exception_decoder.py, so the regex won't match?

Is it possible to control the order in which filters are processed?
If we can't control the order, then modifying the regular expression might help.

In my environment, decoded with option time.
filter_exception_decoder.py line:35
`
r"(?:^\d+:\d+:\d+.\d+ > Backtrace:|^Backtrace:) ?((0x[0-9a-f]+:0x[0-9a-f]+ ?)+)\s*"
`

However, if you use it with options other than time, it might not be decoded.

GOB52, 2021-05-22:
I wrote

`monitor_filters = time, esp32_exception_decoder`

and it worked.

Analyzed how the filter is processed.

https://github.com/platformio/platformio-core/blob/master/platformio/commands/device/helpers.py
line:91
```
miniterm.TRANSFORMATIONS[obj.NAME] = obj
```

https://github.com/pyserial/pyserial/blob/master/serial/tools/miniterm.py
line:352
```
TRANSFORMATIONS = {
    'direct': Transform,    # no transformation
    'default': NoTerminal,
    'nocontrol': NoControls,
    'printable': Printable,
    'colorize': Colorize,
    'debug': DebugIO,
}
```
line:454
```
transformations = [EOL_TRANSFORMATIONS[self.eol]] + [TRANSFORMATIONS[f]
                                                             for f in self.filters]
self.tx_transformations = [t() for t in transformations]
self.rx_transformations = list(reversed(self.tx_transformations))
```
line:505
```
for transformation in self.rx_transformations:
    text = transformation.rx(text)`
```

Except for Miniterm's built-in filters, they seem to be processed in the reverse order of enumeration.

If this behavior is a specification, could you please add a note in the documentation about the processing order when multiple filters are specified in monitor_filters?

(Sorry if there is already a description about the processing order).

theficus, 2022-01-23:
@gob52 I came here to file the same bug as I came across this yesterday. Thanks for finding posting the workaround. At least it's simple enough. 

Hopefully this bug can be fixed, or at the very least the docs can be updated to call out this important detail. I just checked the docs at https://docs.platformio.org/en/latest/core/userguide/device/cmd_monitor.html#cmd-device-monitor-filters and didn't see anything about the ordering of filters being of importance.

closed 2022-10-01 by stale[bot]