platformio/platform-espressif32 #546: Is ESP32-S2 SPIFFS not supported with Arduino Framework?

pattobrien, 2021-05-21:
Hi everyone -

I've been trying to get SPIFFS working on an ESP32-S2-MINI-1 module for the past 24 hours with no luck. Everything builds and uploads fine, but I get the following error after restarting:

```
E (15984) SPIFFS: mount failed, -10025
[ 16439][E][SPIFFS.cpp:114] format(): Formatting SPIFFS failed! Error: -1
[ 16439][E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
```

And here's my platformio.ini file contents:

```
[env:esps2]
platform_packages = 
	framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.0-alpha1
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
board_build.mcu = esp32s2
monitor_speed = 115200
```

Has anyone had any success using SPIFFS with an ESP32-S2 using PlatformIO / Arduino-ESP32 Framework?

valeros, 2021-05-24:
Hi @pattobrien  At the moment there is no stable Arduino package for ESP32-S2.

simon-jouet, 2021-06-20:
Hi @pattobrien, did you by any chance figure a workaround for this?

I'm running in exactly the same issue. Trying the spiffs example with the esp-idf is working fine on the esp32s2 but when setting up platform.io with arduino-esp32 I'm getting a similar error:

```
10:27:19.920 > E (92) SPIFFS: mount failed, -10025
10:27:19.923 > E (93) example: Failed to mount or format filesystem
```

I've tried a few workaround but so far without much success...

pattobrien, 2021-06-24:
@simon-jouet unfortunately no, I wasn't able to figure it out after a couple days of trying. I had my team stop digging, and re-prioritized our efforts in hopes that the upcoming official Arduino 2.0.0 release will trigger PlatformIO support for the S2 and C3.

simon-jouet, 2021-06-30:
Thanks @pattobrien, I'm actually surprised this is related to platformIO, I don't really see why that would be the case. I might try to build the same code using the esp-idf and arduino-esp32 as a component to see if there is any difference

Do you have by any chance the different things you've tested on your hand? Just so I can give it a go as well without repeating what you've done already :)

simon-jouet, 2021-07-03:
@pattobrien Okay I got it working on my end, just letting you know in case it solves the issue for you too.

The problem for me was that my board definition in platformIO was incorrect, my board was set in flash mode dio instead of qio which was the source of the issue. I've modified the board definition to `"flash_mode": "qio"` and it solved it for me.

I noticed this because it was working fine on the Arduino IDE but not on platformIO, after a bit of poking around I noticed that flashing the Arduino bootloader fixed the issue and at this point i realised that the difference between the two environments was the flash_mode

dmitrykez, 2021-07-28:
Since platformIO 3.3.1 release `"flash_mode": "qio"` won't help anymore. 
Any estimation when file system support for ESP32S2 will be implemented?

valeros, 2021-07-28:
@dmitrykez As I already mentioned earlier in this thread, filesystems for S2 will be added once there is a stable Arduino core for this SoC.

sweproj, 2021-08-24:
@dmitrykez 

I have got it to work by set "flash_mode": "qio", in the board file. (same thing as @pattobrien) 
This will break the boot process of the esp32s2. So i need to flash the bootloader seperate after the upload within platformio.
like this esptool.py --chip esp32s2 --port /dev/ttyUSB12 write_flash  0x1000 bootloader_qio_80m.bin, havent figured out why it breaks but anyway

unaiur, 2021-09-08:
Arduino 2.0.0 with support for ESP32-S2 & C3 was released last week.

nickleijenhorst, 2021-12-25:
I'm having the same issue, it's really mindblowing to be honest. I even tried putting flash mode in qio and reflashed bootloader, but still the same issue. What also confuses me, is that the flash mode is set to QIO in Arduino, but when the device actually boots, it says its working in DIO mode. Neither QIO or DIO seems to work for me in PlatformIO, which is quite annoying for my project... I'm now forced to compile with Arduino lol.

ArnieO, 2022-03-25:
Since some of the comments here are getting a bit old (and I believe framework has been updated in the mean time):

Reflashing the "qio" bootloader as described by @sweproj here above solves LittleFS-problem for me: https://github.com/platformio/platform-espressif32/issues/546#issuecomment-904811339

Only flashing with "qio" bootloader does not work.
You need to flash with "dio" bootloader first, then reflash "qio" bootloader.

I use S2 bootloaders from here: https://github.com/espressif/arduino-esp32/tree/master/tools/sdk/esp32s2/bin

closed 2022-04-17 by valeros

diegostf, 2023-03-16:
> Since some of the comments here are getting a bit old (and I believe framework has been updated in the mean time):
> 
> Reflashing the "qio" bootloader as described by @sweproj here above solves LittleFS-problem for me: [#546 (comment)](https://github.com/platformio/platform-espressif32/issues/546#issuecomment-904811339)
> 
> Only flashing with "qio" bootloader does not work. You need to flash with "dio" bootloader first, then reflash "qio" bootloader.
> 
> I use S2 bootloaders from here: https://github.com/espressif/arduino-esp32/tree/master/tools/sdk/esp32s2/bin

How can I make this reflashing? Any tutorial?

Jason2866, 2023-03-16:
@diegostf Easy, use latest Platformio release for espressif32 and flash your device with Platformio ("build and upload"). That's all. All bugs have been fixed. No workarounds needed anymore.

diegostf, 2023-03-17:
> @diegostf Easy, use latest Platformio release for espressif32 and flash your device with Platformio ("build and upload"). That's all. All bugs have been fixed. No workarounds needed anymore.

Tks @Jason2866 .
The problem with  "LittleFS mount failed" was due to not formatting the littleFS on first boot. I used a definition that works with ESP32 and ESP8266 (#define FORMAT_SPIFFS_IF_FAILED true), but it doesn't seem to work with ESP32-S2. So I made a modification to the code. 

```
if(!LittleFS.begin()){
      Serial.println("[ LittleFS ] Formated");
      LittleFS.format();
      if(!LittleFS.begin()){
         Serial.print(F("[ LittleFS ] Error while mounting LITTLEFS\n\n"));
         return false;
      }
   }
```
Now everything works great.

commit 4371944c (2022-04-17): Add support for new filesystems

Resolves #546, resolves #570, resolves #643