platformio/platform-espressif32 #549: Code fails to run when built with platformIO that works through arduine IDE

hspeijer, 2021-05-23:
I am trying to run the following code 

`#include <M5Stack.h>

// the setup routine runs once when M5Stack starts up
void setup() {
  
  // initialize the M5Stack object
  M5.begin();

  /*
    Power chip connected to gpio21, gpio22, I2C device
    Set battery charging voltage and current
    If used battery, please call this function in your project
  */
  M5.Power.begin();
  
}

// the loop routine runs over and over again forever
void loop(){
  // Lcd display
  M5.Lcd.fillScreen(RED);
  delay(1000);
  M5.Lcd.fillScreen(GREEN);
  delay(1000);
  M5.Lcd.fillScreen(BLUE);
  delay(1000);
}`

This works just fine using arduino IDE but it fails when trying to build using platform IO. 

platform.ini:
`[platformio]
src_dir = main

[env:m5-mesh]
platform = espressif32@3.2.0
board = m5stack-core-esp32
framework = arduino
upload_protocol = esptool
upload_port = /dev/ttyUSB0
upload_speed = 460800
monitor_speed = 115200
lib_deps =
  m5stack/M5Stack @ ^0.3.1`

When I monitor the program I get 

`ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
M5Stack initializing...[E][sd_diskio.cpp:194] sdCommand(): Card Failed! cmd: 0x00
[E][sd_diskio.cpp:775] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[E][sd_diskio.cpp:194] sdCommand(): Card Failed! cmd: 0x00
[E][esp32-hal-cpu.c:93] addApbChangeCallback(): duplicate func=400EDB6C arg=3FFBF128
OK
`
I am fairly new to platform IO and esp32. Any help in debugging this issue is much appreciated.

hspeijer, 2021-05-23:
I disabled the SD Card initialization for the M5 which removed the errors but it still does not update the display as expected. I can add debug statements to the loop which show up but the display is not updated. 

As far as I can tell I am using the same versions of esp arduino (1.06) and m5 stack (0.3.1) when I use platformio or arduino ide.

valeros, 2021-05-24:
Hi @hspeijer ! Unfortunately, such kind of issues is very hard to debug when there is no obvious reason to fail. I don't have any M5-based board at hand so I won't be able to help you here. But here are a few steps I would take:

- Make sure the wiring is good and the package versions are indeed the same.
- Completely erase the flash memory via the `pio run -t erase` command and try to upload again.
- Try another libraries that work with the display or an SD card just to make sure there is no problem with the `M5Stack` library.
- Downgrade the version of the platform to `3.1.0` (PlatformIO will download the Arduino package v1.0.5), just to make sure there was no regression introduced in the latest framework package.

hspeijer, 2021-05-24:
I installed the arduino command line tools to see the actual compiler output. I see differences in the number of symbols in the elf file and I think I saw some different compiler options. I can try to see if I can make them point to the same sources. Thanks for your suggestions! The wiring is not something I can check but I did flash the different binary images repeatedly with reproducible results (or lack thereof). Other then the fact that I am having trouble to get it to work they are sweet devices: 
https://docs.m5stack.com/en/core/core2 https://docs.m5stack.com/en/core/basic

closed 2022-08-10 by stale[bot]