platformio/platform-espressif32 #550: Setting app version for espressif32/arduino?

maxbbraun, 2021-05-23:
Is there a way to set the [ESP32 app version](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system.html#app-version) from `platformio.ini` or from some other configuration?

The documentation suggests that this is possible by setting the `PROJECT_VER` variable when using the [`espidf`](https://docs.platformio.org/en/latest/frameworks/espidf.html) framework, but I'm using the [`arduino`](https://docs.platformio.org/en/latest/frameworks/arduino.html) framework.

I tried adding `-DPROJECT_VER=<VERSION>` to `build_flags`, but that doesn't seem to get picked up.

Any ideas?

valeros, 2021-05-24:
Hi @maxbbraun ! 

> I tried adding -DPROJECT_VER=<VERSION> to build_flags, but that doesn't seem to get picked up.

`PROJECT_VER` is an internal variable used by the ESP-IDF's build system, so it's automatically added to the compilation process. In case with Arduino, you need to manually handle it somewhere in your code.

Anyway, could you please elaborate on how you use the `PROJECT_VER` macro in your code?

maxbbraun, 2021-05-24:
Thanks for replying, Valerii!

I was planning to use this indirectly through calls to [`esp_ota_get_app_description()`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/ota.html#_CPPv427esp_ota_get_app_descriptionv) and [`esp_ota_get_partition_description()`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/ota.html#_CPPv433esp_ota_get_partition_descriptionPK15esp_partition_tP14esp_app_desc_t) when managing OTA updates.

The resulting [`esp_app_desc_t.version`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/app_image_format.html#_CPPv4N14esp_app_desc_t7versionE) always seems to have the same value and I would like to be able to set my own version number when building a firmware image.

valeros, 2021-05-24:
It seems that there is no easy way to use that functionality with the Arduino framework. The const instance of the `esp_app_desc_t` struct that contains the application version info is shipped as a precompiled static library with some default values. I suppose I'd need to use higher-level abstractions (e.g. the `Arduino OTA` library) for OTA updates.

closed 2021-05-24 by valeros