platformio/platform-espressif32 #551: Linking fails

brunojoyal, 2021-05-24:
I posted this issue also in platformio/platformio-vscode-ide but it probably wasn't the right place.

On two different machines, I'm suddenly getting this error when I try to compile a blank arduino-esp32 project.

platformio.ini:

```ini
[env]
framework = arduino

[env:esp32]
platform = espressif32

platform_packages = 
    framework-arduinoespressif32  @ https://github.com/espressif/arduino-esp32.git

board = esp32doit-devkit-v1
```

main.cpp:
```cpp
#include <Arduino.h>

void setup(){

}

void loop() {

    
}
```

Getting output:

```txt
Archiving .pio/build/esp32/libFrameworkArduino.a
Indexing .pio/build/esp32/libFrameworkArduino.a
Linking .pio/build/esp32/firmware.elf
/home/bruno/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find crt1-sim.o: No such file or directory
/home/bruno/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find _vectors.o: No such file or directory
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32/firmware.elf] Error 1
```

Many thanks for any help.

valeros, 2021-05-24:
Hi @brunojoyal ! There is no guarantee that that it will compile and link successfully when you're pulling the framework directly from the repository. For example, right now the CI badge in the README shows that the build is broken. You can either use the stable version of the package or lock your package to a specific commit.

closed 2021-05-24 by valeros

brunojoyal, 2021-05-24:
Thanks @valeros for the lightning quick answer. What you said makes a lot of sense. I just removed that line and all is ok now. I'd been getting away with this for surprisingly long.