platformio/platform-espressif32 #557: Adafruit ESP32-S2 Feather wrong specification

Zercerium, 2021-06-05:
Hi,
regarding to the [specs ](https://feathers2.io/) the FeatherS2 has 16MB Flash instead of 4MB and 8MB RAM.
So I think (I'm relatively new and therefore not sure)
`"flash_size": "16MB",` , `"maximum_size": 16777216, `, `"extra_flags": "-DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue",` 
should be correct

[featheresp32-s2.json boardfile](https://github.com/platformio/platform-espressif32/blob/master/boards/featheresp32-s2.json)
Not sure how the size of the SPI RAM can be saved in the board file.
Would be nice if someone could take a look at it

valeros, 2021-06-07:
Hi @Zercerium ! That's two different boards, at the moment there is no support for `UM FeatherS2` . The manifest you're referring to is for `Adafruit ESP32-S2 Feather Development Board` that only has 4MB of flash according to [boards.txt](https://github.com/espressif/arduino-esp32/blob/master/boards.txt#L3253)

Zercerium, 2021-06-07:
Hi @valeros thanks for the reply!
The [manifest](https://github.com/platformio/platform-espressif32/blob/a0025d88ce1d8c871431121a590734903c06aa76/boards/featheresp32-s2.json#L30) is also referring to the 16MB flash version, if you follow the link.
`UM FeatherS2` and `Adafruit ESP32-S2 Feather Development Board` are the same boards.

[boards.txt](https://github.com/espressif/arduino-esp32/blob/master/boards.txt#L3253) should be this board [featheresp32.json](https://github.com/platformio/platform-espressif32/blob/master/boards/featheresp32.json) (boards.txt references an esp32 chip but the manifest an esp32-S2 chip)

I could find an entry for the `Adafruit ESP32-S2 Feather Development Board` in the [board.txt](https://github.com/espressif/arduino-esp32/blob/a618fc1361b08aa968407dd81b7b065cca207f46/boards.txt#L837) after some search 😅

(There is also an [EPS32-S2](https://github.com/espressif/arduino-esp32/blob/a618fc1361b08aa968407dd81b7b065cca207f46/boards.txt#L3761) Version without PSRAM and extended flash but i couldn't find it at the Adafruit store)

valeros, 2021-06-07:
> I could find an entry for the Adafruit ESP32-S2 Feather Development Board in the board.txt after some search

Sorry, my bad, I meant that board `Adafruit Feather ESP32-S2 (no PSRAM)`. Anyway, if you could test your changes locally and make sure the board operates just fine, then we can update the manifest accordingly. 

Thanks in advance.

Zercerium, 2021-06-07:
```
board_build.extra_flags = -DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
board_build.partitions = default_16MB.csv
```
works fine. Also the `ESP.getFreePsram()` command shows 8351975. The only problem is, that the board doesnt connect to usb, also mentioned here [No USB Serial port available #4728](https://github.com/espressif/arduino-esp32/issues/4728) but this isnt a pio bug.
(Tested this with the arduino framework `platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.0-alpha1` is this enough? or espidf required?)

valeros, 2021-06-08:
Great, let's then wait for a stable release of the ESP32 Arduino core, I'll then update the manifest.

UnexpectedMaker, 2021-08-31:
> Hi @valeros thanks for the reply!
> The [manifest](https://github.com/platformio/platform-espressif32/blob/a0025d88ce1d8c871431121a590734903c06aa76/boards/featheresp32-s2.json#L30) is referring to the 16MB flash version, if you follow the link.
> `UM FeatherS2` and `Adafruit ESP32-S2 Feather Development Board` are the same boards.
> 

Just for clarity, Maybe in the definition they are the same board, but they should not be. The FeatherS2 board is not an Adafruit board, it's one of my Unexpected Maker boards. It's not made by Adafruit and is only sold by them as one of my resellers.

if you are looking at fixing the board definition, please also rename it to UM FeatherS2 as that is what it is.

```
"name": "UM FeatherS2 ESP32-S2 Development Board",
"url": "https://feathers2.io",
"vendor": "Unexpected Maker"
```

Jason2866, 2021-11-21:
And another not needed use of `-mfix-esp32-psram-cache-issue`.
All ESP32-S2 do not need this, since there is no faulty revision which is needing.
For ALL ESP32 rev. 3 and newer variants of the ESP32 this is NOT needed too.

valeros, 2022-04-15:
Resolved in https://github.com/platformio/platform-espressif32/commit/84f6d42a38a870cf110d2255eaef7960be57cc8f

closed 2022-04-15 by valeros