platformio/platform-espressif32 #564: esp32-S2 NVS issues

chegewara, 2021-06-16:
For tests i am using esp-idf nvs example with output logs:
```
Opening Non-Volatile Storage (NVS) handle... Done
Reading restart counter from NVS ... The value is not initialized yet!
Updating restart counter in NVS ... Done
Committing updates in NVS ... Done
```
Also with arduino `Preferences` library result is suggesting some issue with NVS.

Here is `platformio.ini` i use to test problem with NVS. 
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
framework = arduino
board = esp32dev

; change microcontroller
board_build.mcu = esp32s2

; change MCU frequency
board_build.f_cpu = 240000000L

platform_packages =
  ; use a special branch
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#master

monitor_speed = 115200
upload_port=/dev/ttyUSB0
board_build.partitions = partitions.csv
```

I confirmed that the same arduino example flashed with arduino IDE using master branch works.

T94T, 2022-03-08:
Is there some progress on this issue?

valeros, 2022-04-18:
Hi @chegewara @T94T ! I cannot reproduce the issue with a Preferences example and the upstream version of the platform. Please confirm it still doesn't work with the following project configuration:

```ini
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
framework = arduino
board = esp32dev
```

closed 2022-08-10 by stale[bot]