platformio/platform-espressif32 #568: Feature Request: support ESP32-C3-DevKitM-1 board

zoumiaoer, 2021-06-23:
Feature Request: support ESP32-C3-DevKitM-1 board

closed 2021-06-23 by valeros

zoumiaoer, 2021-06-24:
It seems that this board hasn't been supported till now , then why this issue has been closed?

zoumiaoer, 2021-06-24:
> It seems that this board hasn't been supported till now , then why this issue has been closed?

@valeros

valeros, 2021-06-24:
The board is available in the development branch with ESP-IDF. You can test using the upstream version:
```ini
[env:esp32-c3-devkitm-1]
platform = https://github.com/platformio/platform-espressif32.git
framework = espidf
board = esp32-s2-kaluga-1
monitor_speed = 115200
```

zoumiaoer, 2021-06-25:
> The board is available in the development branch with ESP-IDF. You can test using the upstream version:
> 
> ```ini
> [env:esp32-c3-devkitm-1]
> platform = https://github.com/platformio/platform-espressif32.git
> framework = espidf
> board = esp32-s2-kaluga-1
> monitor_speed = 115200
> ```

But this board runs with the esp32-c3 soc instead of esp32-s2, will it still works?

valeros, 2021-06-25:
Sorry, my bad, it should be `esp32-c3-devkitm-1` as well:
```ini
[env:esp32-c3-devkitm-1]
platform = https://github.com/platformio/platform-espressif32.git
framework = espidf
board = esp32-c3-devkitm-1
monitor_speed = 115200

```

zoumiaoer, 2021-06-28:
> Sorry, my bad, it should be `esp32-c3-devkitm-1` as well:
> 
> ```ini
> [env:esp32-c3-devkitm-1]
> platform = https://github.com/platformio/platform-espressif32.git
> framework = espidf
> board = esp32-c3-devkitm-1
> monitor_speed = 115200
> ```

Thanks for your reply. I'm wondering if that this board is only supported under the espidf framework but arduino framework?

valeros, 2021-06-28:
At the moment only ESP-IDF. Arduino will be added once there is a stable version of the frameworks that supports this SoC.

zoumiaoer, 2021-06-29:
There is a development release: v2.0.0-alpha1 which support esp32-c3 and esp32-s2 in the espressif/arduino-esp32 page.Is there any ways that I can implement this version of arduino core for the esp32 into the arduino framework so that my sketch for esp32-c3 chip or board could be compiled and uploaded successfully?

thijstriemstra, 2021-08-09:
> At the moment only ESP-IDF. Arduino will be added once there is a stable version of the frameworks that supports this SoC.

Until then I'm seeing this error:

```
PLATFORM: Espressif 32 (3.3.1) > Espressif ESP32-C3-DevKitM-1
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - toolchain-riscv-esp 1.80400.0 (8.4.0)
Error: This board doesn't support arduino framework!
```

commit b073ed84 (2021-06-23): Initial support for ESP32C3

Resolve #568, resolve #497