platformio/platform-espressif32 #570: Add LittleFS and FFat filesystem support

maxgerhardt, 2021-06-24:
Right now the only support filesystem builder tool is SPIFFS via `mkspiffs` 

https://github.com/platformio/platform-espressif32/blob/6c30a4a4b51fa33cf7a45f2b65328a5ad10e75b0/builder/main.py#L167-L168

https://github.com/platformio/platform-espressif32/blob/6c30a4a4b51fa33cf7a45f2b65328a5ad10e75b0/builder/main.py#L196-L208

this should be expanded so that, [as for the ESP8266](https://docs.platformio.org/en/latest/platforms/espressif8266.html#selecting-appropriate-filesystem) other filesystems are supported. Per https://github.com/lorol/arduino-esp32fs-plugin and [Arduino-ESP32](https://github.com/espressif/arduino-esp32/tree/master/libraries), SPIFFS, LittleFS and FFat filesystems are supported. 

For generatinon of the image, analogue to `mkspiffs` and `mklittlefs`, the tool [`mkfatfs`](https://github.com/labplus-cn/mkfatfs) may be usable.

maxgerhardt, 2021-06-24:
For reference, I've created a project at https://github.com/maxgerhardt/pio-esp32-fatfs-example with a modified https://github.com/maxgerhardt/platform-espressif32 that allows using FatFs. The only thing that's missing is that `tool-mkfatfs` gets uploaded and referenced as a package, I've just thrown it into the project folder there.

PierreRambaud, 2022-02-21:
The tool works like a charm! Thanks for sharing!

closed 2022-04-17 by valeros

Algonsi, 2022-05-10:
I work with PlatformIO in VSCode, updated to the latest versions to date, to develop applications with ESP32.
Espressif 32 platform version 4.0.0 says "Added support for LittleFS and FFat filesystems (#546, #570, #643)"
I have version 4.2.0 installed and set 'platform = espressif32' in platformio.ini, but when I try to 'Build Filesystem Image' (or Upload) from the menu provided by the PlatformIO icon, I get the error '""mkffat"" not recognized as an internal or external command' and the image is not builded.
Is this issue fixed?
How could I Build or Upload Filesystem Image with espressif32 platform?
Should I keep using the platform 'https://github.com/maxgerhardt/platform-espressif32' as I have been doing so far?

maxgerhardt, 2022-05-10:
With official support, my platform doesn't need to be used anymore. Try and go into `<home folder>/.platformio/platforms` and remove all `espressif32*` folders, then build your project again with the standard `platform = espressif32` directive. It should download & use the latest 4.2.0 platform. Make sure to follow https://docs.platformio.org/en/latest/platforms/espressif32.html#uploading-files-to-file-system to select the correct filesystem.

If that still does not work, please show your project folder structure + full `platformio.ini`.

maxgerhardt, 2022-05-10:
I think the documentation can be improved on that, see PR linked above ^.

Algonsi, 2022-05-10:
Thank you very much for your fast and accurate help.
You made me realize that I had set 
`board_build.filesystem = ffat`
When I have corrected it to 'fatfs', the problem disappeared.
Thanks again, you do a great job.

dalbert2, 2022-06-17:
I'm using PlatformIO with VSCode and ESP-IDF framework. I've imported the LittleFS component and it works nicely, but I can't register LittleFS at the base path of "/" (I think it's because SPIFFS is registered there).  I didn't see anything in menuconfig that would let me remove SPIFFS; does anyone have the secret decoder ring to do this?  Thanks!

commit 4371944c (2022-04-17): Add support for new filesystems

Resolves #546, resolves #570, resolves #643