platformio/platform-espressif32 #573: Debugging on ESP32-C3

frederictobiasc, 2021-06-25:
Hi,
Setting up the new ESP32-C3 seems to work perfectly using the [upstream version](https://github.com/platformio/platform-espressif32/issues/568#issuecomment-868332222). Thanks for your work!
Is debugging also possible yet? I tried it as explained in [the tutorial](https://docs.platformio.org/en/latest/tutorials/espressif32/espidf_debugging_unit_testing_analysis.html#debugging-the-firmware), but it does not break on my breakpoint in the `app_main()`.

valeros, 2021-06-28:
The board manifest contains all required information, so it should. Unfortunately, I don't have any C3-based board at hand, so I cannot test it.

frederictobiasc, 2021-07-02:
Okay. Right now, I'm able to debug via gdb and eclipse, but I haven't had any success with PlatformIO. If you need someone to test, let me know.

valeros, 2021-07-23:
Hi @frederictobiasc ! Recently, we've updated the OpenOCD package for ESP32, could you please run `pio platform update espressif32` in the IDE terminal and try again? Thanks!

closed 2022-09-20 by stale[bot]