platformio/platform-espressif32 #583: New project code blank

Deepblue1996, 2021-07-10:
I create a new project code blank
[image]

Deepblue1996, 2021-07-10:
The project looks like this
[image]

Deepblue1996, 2021-07-10:
Change it to JLink and you can download it
[image]

Deepblue1996, 2021-07-10:
But if I add a header file, I get an error, so how do I add my own code
[image]

valeros, 2021-07-12:
Hi @Deepblue1996 ! At the moment the integration with CLion doesn't support ESP-IDF, see https://docs.platformio.org/en/latest/integration/ide/clion.html#clion-with-esp-idf

closed 2021-07-12 by valeros