platformio/platform-espressif32 #584: Wrong flash_mode used with esptool.py for upload

Jason2866, 2021-07-10:
PlatformIO Core, version 5.2.0a8 used with Windows10 (latest version)

Using this custom board definition [esp32-m5core2.json](https://github.com/platformio/platform-espressif32/files/6795014/esp32-m5core2.json.txt) is merged after successfull compile from esptool.py to this bin.
```
"c:\python37\python.exe" "C:\Users\Hans\.platformio\packages\tool-esptoolpy\esptool.py" --chip esp32 elf2image --flash_mode qout --flash_freq 80m --flash_size 16MB -o .pio\build\tasmota32-core2\firmware.bin .pio\build\tasmota32-core2\firmware.elf
esptool.py v3.1
```
the flashing command (target "upload") is doing that
```
"c:\python37\python.exe" "C:\Users\Hans\.platformio\packages\tool-esptoolpy\esptool.py" --chip esp32 --port "COM3" --baud 2000000 --before default_reset --after hard_reset write_flash -z --flash_mode dout --flash_freq 80m --flash_size detect 0x1000 C:\Users\Hans\.platformio\packages\framework-arduinoespressif32@src-f3d444366f5de6662e203906cb295620\tools\sdk\bin\bootloader_qout_80m.bin 0x8000 D:\Tasmota\.pio\build\tasmota32-core2\partitions.bin 0xe000 C:\Users\Hans\.platformio\packages\framework-arduinoespressif32@src-f3d444366f5de6662e203906cb295620\tools\partitions\boot_app0.bin 0x10000 .pio\build\tasmota32-core2\firmware.bin
```
The `flash_mode` is wrongly `dout` the used bootloader would be correct `bootloader_qout_80m.bin` but does not match to the `flash_mode` used. 
Checking the active mode in the app (Tasmota) shows `"FlashMode":3` which is `DOUT`

valeros, 2021-07-23:
Hi @Jason2866 ! The [code](https://github.com/platformio/platform-espressif32/blob/637992878cda2cd1cf62febc26d4f897fdbeea0a/builder/main.py#L34-L40) responsible for this functionality is 3 years old, so I guess something might have changed since that time and probably esptool.py now supports much more flash modes.

Anyway, what's the correct behavior should be? Does it work with `--flash_mode qout` ?

Jason2866, 2021-07-24:
> Hi @Jason2866 ! The [code](https://github.com/platformio/platform-espressif32/blob/637992878cda2cd1cf62febc26d4f897fdbeea0a/builder/main.py#L34-L40) responsible for this functionality is 3 years [...]

Yes, it does work using `--flash_mode qout`
The flash command used is
```
esptool.py --chip esp32 --baud 2000000 --before default_reset --after hard_reset write_flash -z --flash_mode qout --flash_freq 80m --flash_size detect 0x1000 bootloader_qout_80m.bin 0x8000 partitions_spiffs_10M.bin 0xe000 boot_app0.bin 0x10000 tasmota32-core2.bin
```
Serial log
```
ets Jul 29 2019 12:21:46
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QOUT, clock div:1
load:0x3fff0030,len:324
load:0x40078000,len:13096
load:0x40080400,len:3552
entry 0x400805d0

00:00:00.001 HDW: ESP32-D0WDQ6-V3

00:00:00.006 UFS: FlashFS mounted with 10248 kB free
```
Tasmota code tells:
` ProgramFlashSize":16384,"FlashSize":16384,"FlashFrequency":80,"FlashMode":1 `
FlashMode 1 is QOUT

So, yes `--flash_mode qout` would be the correct change

valeros, 2021-09-04:
Hi @Jason2866 ! I'm working on the integration of the latest Arduino core v2.0 and bumped into this issue again. It seems that the behavior when `qout` is replaced by `dout` was actually consistent with Arduino:
```
m5stack-core-esp32.menu.FlashMode.qio=QIO
m5stack-core-esp32.menu.FlashMode.qio.build.flash_mode=dio
m5stack-core-esp32.menu.FlashMode.qio.build.boot=qio
m5stack-core-esp32.menu.FlashMode.dio=DIO
m5stack-core-esp32.menu.FlashMode.dio.build.flash_mode=dio
m5stack-core-esp32.menu.FlashMode.dio.build.boot=dio
m5stack-core-esp32.menu.FlashMode.qout=QOUT
m5stack-core-esp32.menu.FlashMode.qout.build.flash_mode=dout
m5stack-core-esp32.menu.FlashMode.qout.build.boot=qout
m5stack-core-esp32.menu.FlashMode.dout=DOUT
m5stack-core-esp32.menu.FlashMode.dout.build.flash_mode=dout
m5stack-core-esp32.menu.FlashMode.dout.build.boot=dout
```
And if I upload from the Arduino IDE, I see the exactly same command with `dout`:
```
C:\Users\vbox\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.1.0/esptool.exe --chip esp32 --port COM5 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 C:\Users\vbox\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0/tools/partitions/boot_app0.bin 0x1000 C:\Users\vbox\AppData\Local\Temp\arduino_build_747943/Blink.ino.bootloader.bin 0x10000 C:\Users\vbox\AppData\Local\Temp\arduino_build_747943/Blink.ino.bin 0x8000 C:\Users\vbox\AppData\Local\Temp\arduino_build_747943/Blink.ino.partitions.bin 
```
As far as I know, there were no complains from users about this, except your case. Any thoughts?

Jason2866, 2021-09-04:
True, looked in boards.txt. But when using mode QIO i never had a starting device.
Just did a short test with Arduino IDE. It is booting and using flash mode QIO!

Jason2866, 2021-09-04:
Hi @valeros again,
i did some research the behaviour before the change is correct https://github.com/espressif/esptool/wiki/SPI-Flash-Modes#how-is-flash-mode-communicated-to-the-esp8266esp32
But something has to be different in the final code that initiates the QIO mode in Arduino IDE.

valeros, 2021-09-06:
The only obvious difference between upload commands is flash frequency, in PlatformIO manifest it's set to 40MHz while in Arduino it's 80MHz. I'm inclined to reverse the commit that fixed this issue, but unfortunately, I don't have this board to test it out, so I'd be great if you could experiment with upload commands to resolve the problem in a proper way.

Jason2866, 2021-09-06:
@valeros i think i find time to do tests this evening (europe time zone). The flash speed should not make the difference. Anyway i try this too.

Jason2866, 2021-09-06:
@valeros I think i know now. The way it was done before is wrong and as it is now is wrong too. With Arduino IDE you select the mode in which the flash is working! when the sketch is running (example ESP32 https://github.com/tasmota/arduino-esp32/blob/master/boards.txt#L428-L444)
```
esp32.menu.FlashMode.qio=QIO
esp32.menu.FlashMode.qio.build.flash_mode=dio
esp32.menu.FlashMode.qio.build.boot=qio
```
and all needed modes are choosen accordingly.
With Platformio the  Flash mode is defined in boards.json and flash_mode is fixed defined.
ESP32: "flash_mode": "dio" https://github.com/platformio/platform-espressif32/blob/develop/boards/esp32dev.json#L10

There is no possibility to do the build variant:
flash_mode: qio
build.flash_mode=dio
build.boot=qio

correctly, since a change from the entry to `qio` builds a wrong firmware with set magic byte `qio` instead of `dio`
Just doing the correct esptool.py command sequence *after* the build is not working. The magic byte for the flash mode in the firmware is wrong set!
It is the 3rd byte in the firmware binary. The byte after "E9 05" defines the flash mode of the firmware:
"00" = QIO
"01" = QOUT
"02" = DIO
"03" = DOUT

So the solution is, revert the change, and patch the firmware (3rd byte) with the same logic
 replace "00" with "02"
 replace "01" with "03"

Jason2866, 2021-09-06:
After doing more tests, the way it is now works. But it is not inline with Arduino.

valeros, 2021-09-06:
That's interesting, because before the commit that resolved this issue, the behavior was consistent with Arduino and flash mode used in the final upload command was correct:
```
def _get_board_flash_mode(env):
    mode = env.subst("$BOARD_FLASH_MODE")
    if mode == "qio":
        return "dio"
    elif mode == "qout":
        return "dout"
    return mode
```

I will double check the flags before rolling out the update for Arduino v2.0.

Jason2866, 2021-09-06:
Have you ever tried to build a firmware in mode QIO? Never got a working version with Platformio.

Jason2866, 2021-09-09:
@valeros Please revert the change. Regarding to the esptool.py wiki this is correct. https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#software-bootloader-header-info
If the mode QIO is not activated (at runtime) is a issue in the software bootloader which enables Quad I/O based on the detected flash chip mode.

commit fe5d2796 (2021-07-26): Use default flash mode when uploading via esptoolpy

Resolve #584
commit f7bcd7e6 (2021-11-05): Revert "Use default flash mode when uploading via esptoolpy"

Resolve #584, resolve #630, resolve #623

This reverts commit fe5d2796d46cc4fb29f8a77cbfa66f53faf4fb89.
closed 2021-11-05 by valeros