platformio/platform-espressif32 #587: Problem with SD.begin() function with new version 3.3.0

florian-fouquet, 2021-07-22:
Hi,

I'm using arduino-esp32 as a component and the SD library.
Since version 3.3.0 of espressif32 platform, i have the following error during execution. All was working well before (With version 3.2.0).
[image]

After analysis with scope, i didn't have correct signal on the MISO pin.

After some search, i found people having trouble with SD card who just add a 10K pull-up on the MISO pin.
I first try to put the MISO pin in INPUT_PULLUP (Internal pull-up 45k of the esp32) and then add an external 10K and got the same error during execution :
[image]

I spend 2 days trying to fix the bug without succes.

I finally move back to version 3.2.0 which is working correctly.

valeros, 2021-07-23:
Hi @florian-fouquet ! According to the v3.3.0 notes there shouldn't be anything critical, except that that sdk configuration files are now generated for each environment separately and this feature might not be support by ESP-IDF v4.0.1 that is used when Arduino is used as a component. What is your build log when you compile using v3.3.0?

valeros, 2022-04-18:
Closing as no further info provided.

closed 2022-04-18 by valeros