platformio/platform-espressif32 #588: Regenerate sdkconfig.<board> if misisng

maxgerhardt, 2021-07-22:
I just wanted to apply a new `sdkconfig.defaults` file (from https://github.com/espressif/esp-idf/tree/v4.3/examples/bluetooth/bluedroid/classic_bt/a2dp_sink/main), but I already had a `sdkconfig.pico32` in my project folder after an initial build. So, wanting to apply the defaults file, I deleted `sdkconfig.pico32` in hopes that the next `pio run -t menuconfig` regenerates the default options plus the options from the `sdkconfig.defaults` -- instead it gives an assertion error that the file is not found

```
>pio run -t menuconfig
..
Reading CMake configuration...
AssertionError: Missing auto-generated SDK configuration file `C:\Users\Max\temp\espidf_a2dp_sink\sdkconfig.pico32`:
  File "C:\Users\Max\AppData\Local\Programs\Python\Python38\Lib\site-packages\platformio\builder\main.py", line 178:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Max\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Max\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Max\.platformio\platforms\espressif32\builder\main.py", line 228:
    target_elf = env.BuildProgram()
  File "C:\Users\Max\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Max\AppData\Local\Programs\Python\Python38\Lib\site-packages\platformio\builder\tools\platformio.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\Max\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Max\AppData\Local\Programs\Python\Python38\Lib\site-packages\platformio\builder\tools\platformio.py", line 128:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\Max\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Max\AppData\Local\Programs\Python\Python38\Lib\site-packages\platformio\builder\tools\platformio.py", line 343:
    SConscript(env.GetFrameworkScript(f), exports="env")
  File "C:\Users\Max\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 654:
    return method(*args, **kw)
  File "C:\Users\Max\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Max\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Max\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 1164:
    assert os.path.isfile(SDKCONFIG_PATH), (
========================================================================================== [FAILED] Took 2.02 seconds ==========================================================================================
```

Using latest Espressif 32 (3.3.0).

valeros, 2021-07-22:
Thanks for reporting, should be fixed in the dev branch!

closed 2021-07-22 by valeros

maxgerhardt, 2021-07-22:
I've just tested this and the regeneration, while respecting the `sdkconfig.defaults` file, works like a charm. Thanks!

commit 63799287 (2021-07-22): Updated sdkconfig files should trigger CMake to regenerate project

Resolve #588