platformio/platform-espressif32 #589: Serial.Begin() hangs if RX line has traffic at boot in platforms > 3.0.0

mbilsky, 2021-07-22:
I am having a pretty interesting bug that I'm not sure how to fix. If we boot up our code with the serial RX line busy, the ESP32-Wrover hangs. After some hardware debugging (testing on the ESP-WRrover-kit) the issue seems to be happening in Serial.begin(). More specifically, after uartAttachRx(uart, rxPin, inverted) is called in esp32-hal-uart.c . We didn't have this issue in earlier versions of the IDF (we're using Arduino on platformIO and moved from 3.0.0 to 3.X.0). 3.1-3.3 all seem to have the same problem. We can bodge around it by being careful with the other MCUs the ESP32 is talking to but I'd rather just fix the bug :) 

It appears that the RX interrupt is firing continuously preventing other code from executing.

I've tried initializing Serial in void loop (in a run once block) to confirm that we can make it through the rest of setup just fine. Same issue happens.

I can also successfully run: ` Serial.begin(115200,SERIAL_8N1,-1,1); ` in void setup() to get serial output just fine. Then end the serial and restart with normal ` Serial.begin(115200) ;`

Any thoughts? Thanks!

valeros, 2021-07-23:
Hi @mbilsky ! Does it work as expected in Arduino IDE?

mbilsky, 2021-07-23:
Hey @valeros, great troubleshooting idea!

I tested in Arduino 1.8.13. Here are the results by package version:

1.0.6 -> hangs on boot
1.0.5 -> hangs on boot
1.0.4 -> No issues!

Ordinarily I would just stay at 3.0.0 that uses version 1.0.4 of the Arduino framework but we are having issues where our camera isn't giving us the full 1600x1200 unless we use version  >=3.1.0

valeros, 2021-07-23:
It seems that the issue is hidden somewhere in the Arduino core for ESP32 and has nothing to do with PlatformIO. I'd recommend submitting an issue to the Arduino repository here https://github.com/espressif/arduino-esp32 so a bug fix can land in the upcoming core v2.0.

mbilsky, 2021-07-23:
Thanks! I'll close the issue here and update it when I find a fix

closed 2021-07-23 by mbilsky

mbilsky, 2021-07-24:
For anyone else with the issue I have posted on the Arduino core github here: https://github.com/espressif/arduino-esp32/issues/5443#issuecomment-885775959

Figured out a temporary work-around and we'll see if it gets fully fixed in v2.0+